raspberrypi/documentation #674: Need command-line methods for Zero

hariedo, 2017-06-24:
On the page at https://www.raspberrypi.org/documentation/remote-access/vnc/ we see this paragraph:

> If you're connecting from a non-RealVNC Viewer app, you'll first need to downgrade VNC Server's authentication scheme, specify a password unique to VNC Server, and then enter that instead. To do this, open the VNC Server dialog on your Raspberry Pi, select Menu > Options > Security, and choose VNC password from the Authentication dropdown.

This cannot be done on a headless Zero. What file contains the VNC Server's authentication settings?  Can this be done with a little 'sudo nano' editing?  There seems to be nothing in the default /etc/vnc/config or /etc/vnc/config.d/common of use here.

Also worth a mention, the MacOS built-in VNC viewer is a non-RealVNC Viewer, so this is a pretty typical scenario. Mentioning MacOS in that sentence may help when googling.

waveform80, 2017-06-28:
There's some command line tools which you may find useful: `vncinitconfig` and `vncpasswd`. They've both got man-pages and the former (which is responsible for setting up various RealVNC configuration files) is a simple bash script (I haven't read through it to see where specific things are but it's probably worth a browse).

JamesH65, 2017-07-10:
@minkustree Can RealVNC comment on this one please?

jamleed, 2017-07-11:
Hopefully the command you need is:

sudo vncpasswd -service

This will prompt you to set a password and insert it for you in the right config file for VNC Server running in Service Mode, which is the default configuration on the Pi.

To downgrade the authentication scheme to VNC password (from the default Linux system auth), open the /root/.vnc/config.d/vncserver-x11 config file and replace 'Authentication=SystemAuth' with 'Authentication=VncAuth'. Then, restart VNC Server.

There are several other useful commands on the following page of our website: https://www.realvnc.com/docs/deployment-script.html

Please be aware some only apply to our Professional and Enterprise subscriptions. Hopefully these are clearly marked.

JamesH65, 2017-07-11:
I'd be happy to accept a PR on the documentation if you think this should be in there. I think the RealVNC page was actually written by the RealVNC people in the first place.

jamleed, 2017-07-11:
Thanks. Yes, it was written by me. I'll see if I can find some time to create one!

PR #707 (2017-09-15): VNC password command-line config instructions
Added command-line instructions for configuring a VNC password for
authentication.

Fixes #674
commit ca45a431 (2017-09-15): VNC password command-line config instructions (#707)

* VNC password command-line config instructions

Added command-line instructions for configuring a VNC password for
authentication.

Fixes #674

* copy edits
closed 2017-09-15 by JamesH65