raspberrypi/documentation #676: Compute Module 3 IO Board's Design Files

motters, 2017-06-26:
Hi,

We've noticed that you provide the CMIO board's design files for revision 1.1 and 1.2 in the documentation but you don't provide the design files for revision 3.0.

Is it possible for you to upload these files or are you not providing them by default? 

The schematic is available but we would just like to wiz around the PCB layout before designing our custom mother board so to speak.

Sorry if this has been asked before i couldn't see any open or closed issues relating to this.

Thanks

Sam

JamesH65, 2017-06-27:
I've checked up on this, it's in the pipeline to do, just not had time yet. Not sure of schedule, we a re a bit busy at the moment.

JamesH65, 2017-10-17:
I think these are now on the website. Please reopen with reasoning  if there are still things missing.

https://www.raspberrypi.org/documentation/hardware/computemodule/designfiles.md

closed 2017-10-17 by JamesH65