raspberrypi/documentation #682: Guidance for using systemd to start services at boot

buxtonpaul, 2017-06-29:
There has been some discussion on the forums on more guidance on starting scripts at boot. The current guidance is to use rc.local but as the documentation points out, if this is misconfigured you can stop your pi booting. I have (before realising I should file an issue first) prepared some guidance on using systemd instead, which has the benefit of being more robust to misconfiguration and is more controlable once setup.
This is related to pull request #681. Apologies for doing things in the wrong order!
Paul

JamesH65, 2017-06-29:
Looks good, thanks. I need to check it technically, then send to copy
editors, so won't get merged straight away.

bennuttall, 2017-06-29:
Excellent - thanks for the PR. Will be great to have this in the docs!

JamesH65, 2017-08-03:
I've not had time to tech check this, and will be out of office for the next couple of weeks. Sorry about the delay.

JamesH65, 2017-09-07:
Merged with #681

closed 2017-09-07 by JamesH65