raspberrypi/documentation #686: Error : Setting up a Raspberry Pi as an access point in a standalone network

julienpltr, 2017-07-04:
Hello,

I try to make a Wifi access point with my rpi3 but by following the [website documentation](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md#internet-sharing)

> Add a new bridge, which in this case is called br0.
`sudo brctl addbr br0`

> Connect the network ports. In this case, connect eth0 to wlan0.
` sudo brctl addif br0 eth0 wlan0`

I get this output : 

> can't add wlan0 to bridge br0

Actually, I understand that kernel doesn't allow to bridge ethernet and wlan.
The old documentation worked very well, is that possible to change documentation in the right way ?

Cordialy

JamesH65, 2017-07-07:
Not sure what the problem is, those instructions work fine for everyone else...you can clearly bridge from eth0 to wlan0, there is no kernel limitation there.

JamesH65, 2017-07-10:
@julienpltr Do you still see this issue? I'm at a loss to suggest what the problem might be and I don;t want to change the docs until we know specifically what the issue is.

julienpltr, 2017-07-19:
Sorry for the late response,
I try again with a fresh install of Raspbian on my Rpi3, it still doesn't work.

I will search again in the coming days.

hassanbot, 2017-07-20:
I'm having the same issue, with Raspbian Jessie Lite July 2017 (released 2017-07-05). Getting this error message:
```
pi@raspberrypi:~ $ sudo brctl addif br0 eth0 wlan0
can't add wlan0 to bridge br0: Operation not supported
```

hassanbot, 2017-07-20:
It worked for me by just skipping the `sudo brctl addif br0 eth0 wlan0` line in the tutorial. I also did a reboot after completing the tutorial, but I'm not sure that's necessary.

alt55, 2017-08-24:
Could it be that something has changed in the latest releases (august 2017) in Noobs or in Jessie? I followed the description to the letter, it starts to go wrong  from step one.

JamesH65, 2017-08-24:
I suspect that quite a few of these documents will need to be re-written for the latest release. The whole of networking has had changes, and now works using a different configuration system. I really don't have time right now to investigate further though.

alt55, 2017-08-24:
I suspected that. Thanks anyway

JamesH65, 2017-09-07:
There is an explanation of the error and why it happens here. https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=192355&p=1206657&hilit=sudo+brctl+addif+br0+eth0+wlan0#p1206326

The docs will be updated after confirmation.

julienpltr, 2017-11-01:
Hello !
I follow the new step in documentation and now it's working !! Thank you so much !!

closed 2017-11-01 by julienpltr

skiddaw, 2017-11-04:
Julien,
Do you mean the documentation in GitHub or the documentation in RaspberryPi.org or are they the same thing?
Is it true this doesn't work for Stretch-Lite and should that be explained in the documentation.
Should perhaps documentation be given a number and revision so we all know where it's at.  There is nothing on Ras...Pi.org to say it's been changed.
Should perhaps documentation itemize the working operating system(s) and other requirements before guys like me dig themselves into a hole loading Lite etc.
My thanks to all who have participated in a solution, my comments are more for the overall framework of documentation not this particular case.  I'd suggest a boiler plate which sets out a level playing field.