raspberrypi/documentation #693: Mac dd command needs suffix conv=sync

rainerstandke, 2017-07-13:
If it doesn't have that then apt-get upgrades will fail. I am not enough of a Linux expert to just edit this, but I think you guys should do that, append it to all dd commands on this page.

This is what clued me in:

https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=187936&p=1186098#p1186098

XECDesign, 2017-07-13:
`conv=fdatasync` is the correct option, but I'd suggest that the update for the mac instructions should mirror https://github.com/raspberrypi/documentation/pull/692 and use `; sync` instead.

rainerstandke, 2017-07-13:
Well, I did try that and got:

dd: unknown conversion fdatasync

XECDesign, 2017-07-13:
Ah.. Mac OS and its slightly different set of tools.

rainerstandke, 2017-07-13:
But since this page seems to be precisely about that - i.e. the Mac - it does make sense to use my suggestion, no?

XECDesign, 2017-07-13:
I don't know enough about the internal's of that version of dd. Is it just rounding down the final block instead of padding it out? If that's what it's doing, then your suggestion makes sense. The manpage is unclear. To avoid ambiguity, I recommend using `sync` as a separate command instead, assuming it's available and works the same way on macs.

rainerstandke, 2017-07-13:
My point is this page should be updated to save some folks some time and grief. 

And with that, I rest my case.

P.S. Thanks to all who contribute, it's a great resource!

JamesH65, 2017-07-13:
Happy to accept PRs on the docs, we get quite a few community changes
through that route.

XECDesign, 2017-07-14:
To correct myself a bit, @rainerstandke's solution is probably right. Mac OS has the buffered and unbuffered way of referring to block devices, so `sync` shouldn't have any effect when using the unbuffered mode. In that case `conv=sync` is probably the correct solution.

PR #695 (2017-07-17): installing-images: dd - use conv=fsync and conv=sync for Linux and Mac
See https://github.com/raspberrypi/documentation/issues/693 and https://github.com/raspberrypi/documentation/pull/692#issuecomment-315710731

Fixes #693
closed 2017-07-17 by JamesH65