raspberrypi/documentation #700: Audio select documentation is out of date

spl237, 2017-07-28:
The documentation at https://www.raspberrypi.org/documentation/configuration/audio-config.md is out of date - it refers to the output switching in raspi-config (which still works), but doesn't mention the ability to switch the audio output from the volume plugin on the desktop.

JamesH65, 2017-09-07:
@spl237 Do you have time to update, or can you send me the details and I can do it?

JamesH65, 2017-11-20:
@spl237 Any news?

spl237, 2017-11-20:
I'd suggest changing where it says "There are two ways of setting the audio output" to "There are three ways...", and insert the following paragraph before the "Command Line" section:

Desktop Volume Control

Right-clicking the volume icon on the desktop taskbar brings up the audio output selector; this allows you to select between the internal audio outputs. It also allows you to select any external audio devices, such as USB sound cards and Bluetooth audio devices. A green tick is shown against the currently-selected audio output device - simply left-click the desired output in the pop-up menu to change this. The volume control and mute operate on the currently-selected device.

JamesH65, 2017-11-20:
PR at #770

JamesH65, 2018-02-19:
A fix was merged some time ago....

closed 2018-02-19 by JamesH65