raspberrypi/documentation #710: x86 image documentation

bennuttall, 2017-08-14:
Now that we have the RPD ISO on the downloads page, it would be good to be able to link to some documentation on how to use it. I propose:

- How to write the image to a CD/DVD/USB stick
- How to boot from a live disc
- How to install it on a PC
- How to boot it in a VM

I'd be happy to write the VM tutorial, and I know we've had magpi articles on writing to a USB stick, but we should place them in here and keep them up-to-date. I haven't seen any tutorials on how to install it for good - I know there are some new options on boot in the more recent image, and these good do with explaining too.

Any volunteers?

lurch, 2017-08-14:
`these good do with explaining too` -> `these could do with explaining too` ? ;-)
(feel free to delete this comment after fixing)

lurch, 2017-08-14:
Is https://www.raspberrypi.org/downloads/raspberry-pi-desktop/ supposed to say "June 2017"? It currently says
```
Version: June 2016
Release date: 2017-06-22
```
which doesn't seem right?!

bennuttall, 2017-08-15:
Thanks for the comments - have corrected the date on the web page.

JamesH65, 2018-02-19:
Can this be closed? @bennuttall @lurch

JamesH65, 2018-07-02:
@bennuttall Did this get done?

JamesH65, 2018-09-12:
@bennuttall Ping

aallan, 2020-06-03:
This has been open since 2018. Is it still relevant, can we close?

JamesH65, 2020-06-03:
We still need some minimal docs on the x86 Desktop stuff, but it could be as simple as how to install, since in most respects it's the same as Debian. 

Perhaps we should close this one and be a bit more specific on the requirements in a new issue.

bennuttall, 2020-06-03:
There's a PR for it at #711 but see my last comment: https://github.com/raspberrypi/documentation/pull/711#issuecomment-454876720

lurch, 2020-06-04:
Now that "Raspbian" has been renamed to "Raspberry Pi OS", will the new version of "Raspberry Pi Desktop" also be given a different name, or will that be staying the same? Might be worth delaying this issue a bit longer until after the next release of the x86 Desktop, if that's going to be soon?

ping @XECDesign

XECDesign, 2020-06-05:
I believe it will have a different name, but I'm not sure how soon it will be updated, given that a lot of focus in on the arm64 release right now.

aallan, 2021-06-07:
Linked pull request closed, so closing.

closed 2021-06-07 by aallan