raspberrypi/documentation #715: Wireless with static and DHCP

Lewiscowles1986, 2017-08-21:
Not sure where this should go, but 

https://github.com/raspberrypi/documentation/blob/b8d7675196936b2fdcb47ee0619f2e6b170e474f/configuration/wireless/access-point.md will lead to the stretch images not loading any interfaces not defined in `/etc/network/interfaces` which is a real pain :sob: 

https://t.co/Uyljq09Znq & https://gist.github.com/Lewiscowles1986/390d4d423a08c4663c0ada0adfe04cdb fixes this using the -Z command line switch to `dhcpcd` providing a built regex. 

I personally am not sure that the initial check was strong enough, to protect users, but it's in a var in that file, so when it's updated, so is the grep it does. That also helps stop the line-wrap.

This would be something I'd like in a huge callout anywhere in docs there is a suggestion to edit `/etc/network/interfaces`

I would also like to know if there is a way to provision static interfaces via the dhcp and see that linked in docs.

Let me know if you fancy a PR, or if you know where I can PR the edit to dhcpcd :wink:

JamesH65, 2017-09-07:
Is there a better way to do the whole thing in Stretch that avoids this sort of thing? A revamp of the entire AP docs?

Lewiscowles1986, 2017-09-07:
What is it you'd like to avoid, a patch, or blocking off interfaces?

As I see it, the changes made for stretch by raspberrypi team broke how a linux system works, but doesn't give any indication to users this would happen, or how to replicate functionality for scripts that used to use `/etc/network/interfaces`

I'm personally happy to plod along with my script, but others may benefit from updated docs. I'm open to ideas on the subject. Clearly the changes were made for a reason

JamesH65, 2017-09-08:
All these changes came from upstream, nothing to do with the RPF. So they are real Linux. My preference is docs that match the SW, rather than using scripts. When I have time I'll try to figure out the correct docs.

Lewiscowles1986, 2017-09-08:
I'm sorry, I don't want to get off topic or into arguments, but I'm unable to reconcile `/usr/lib/dhcpcd5/dhcpcd` with the stretch distro for armhf or amd64, and heard that it was a RPF change from the IRC channel. I've also downloaded http://ftp.de.debian.org/debian/pool/main/d/dhcpcd5/dhcpcd5_6.0.5-2_armhf.deb and cannot find the file in the source deb, so it seemed to support what I've said. Rather than get caught up in that, lets say for the rPi platform it's changed between jessie when released, and stretch.

I am aware the interface names have changed in linux in general; I'm not complaining about the name of interfaces (I know I can pass kernel flags if I want to alter that, but would also like to see that documented too). It's more-so the fact that the default behaviour is to not bring up ***ANY*** interfaces if even one is defined in `/etc/network/interfaces`, which I can't replicate for other devices on mainline debian armhf or amd64.

Yesterday I attended a presentation by my local LUG and one of our members was frustrated, unable to get network interfaces to come up. This person is very capable with networks, they manage the infrastructure for the LUG and used to be chairperson. I modified the file from the gist I've created, and it worked for them. 

This is why I am suggesting updating docs. Alternatives I'd be equally as happy to accept are changes/reversions of that script; alternative scripts, forward porting from jessie; although as mentioned I'm happy to hear of more alternatives. My main concern is for users, and keeping software simple for them.

JamesH65, 2017-09-08:
@Lewiscowles1986 Hmm, I was thinking more of software changes than configuration, so my bad on that one. We may well have config changes, but I have not idea what may have changed and why.

@XECDesign @pelwell @spl237  Have you guys any input on this? I know nothing about the changes to networking in Stretch.

XECDesign, 2017-09-08:
I have already explained this to OP on IRC, but for anybody else who might stumble upon this...

The Debian package has always had a similar check, but it was taken out in Jessie when we added a systemd unit for it. Before, it was launched through an init.d script with the following check:
```bash
	if grep -q "^[[:space:]]*iface[[:space:]]*.*[[:space:]]*inet[[:space:]]*dhcp" \
	$INTERFACES; then
		log_failure_msg "Not running $NAME because $INTERFACES"
		log_failure_msg "defines some interfaces that will use a" \
			"DHCP client"
		exit 6
	fi
```
You can see this in the .deb OP linked. This behaviour was re-introduced in our Stretch package, because it's the right thing to do.

You can look on the forum to find countless threads of people who are not aware that dhcpcd is a thing and just want to configure their network using ifupdown (/etc/network/interfaces). They found that dhcpcd would run anyway and undo their changes.

dhcpcd and ifupdown were not designed to be aware of each other and the user should use one or the other, not randomly configure one interface through one system and the rest through another. 

The OP's solution for this is to only run dhcpcd for interfaces not configured through ifupdown, which seems like a very good one, but I am a little cautious about picking up. The main reason that I haven't added this to the dhcpcd package is that I don't know what potential problems this might introduce in more exotic cases and have no interest in finding out.

Lewiscowles1986, 2017-09-08:
Where is this in the deb I linked? (it's not in the same place as on the Pi)

While I accept you don't want the script I've supplied, what solutions will there be for users? All my script does is use the `-Z` flag of `dhcpcd` which is a core part of it's programming; repurposing the regular expression / regex-like pattern from what we're now saying is Debian. 

I'm not experiencing an exit or the file in Debian stretch on amd64 vm or any other arm devices.

XECDesign, 2017-09-08:
Not in the same place, but in `/etc/init.d/dhcpcd'.

The proposed solution will be to configure the network though dhcpcd or through ifupdown. I like what you've done, but I don't want to be in the situation of having to troubleshoot complicated frankenstein network configs or set the expectation that dhcpcd, ifupdown, wicd, NetworkManager and everything else should just work together when they don't on any other system.

For example, if you look at this page:
https://wiki.debian.org/WiFi/HowToUse#Wicd

You see that they tell you to uninstall NetworkManager and clear /etc/network/interfaces.

I don't want to deviate from Debian's behaviour for this, but it's not just up to me. Maybe I am being overly cautious, so if @pelwell , @spl237 and @JamesH65  are happy with this, I am happy to add it in.

Lewiscowles1986, 2017-09-08:
Again, not looking to push this (I'd rather not have to maintain a thing either). I have always wanted another solution, but the script is for the headless "lite" edition.

I know the franken-build worries, so I'm equally as willing to avoid. If we have a suggestion for how to achieve the same (perhaps using dhcpcd5) I'd be equally as happy with that, but think it should live in official docs.

aallan, 2020-06-03:
Open since 2017 waiting for external comment. Can we close?

JamesH65, 2020-06-03:
I suspect this is all out of date now anyway, now we are on Buster. So closing. 

Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2020-06-03 by JamesH65

Lewiscowles1986, 2020-06-04:
It's still needed for any distro using dhcpcd AFAIK if you touch /etc/network/interfaces a patch I distribute via gist is still necessary.

lurch, 2020-06-04:
> The core use case for this is setting a single interface to be static, while maintaining DHCP on others

How common a use-case is that? Can you give an example of when / where / why that'd be useful?

Lewiscowles1986, 2020-06-05:
access points & headless BYOS (Bring your own screen) kiosks
network bridges
traditional device WiFi enablement bridges
most multi-network-spanning use cases

lurch, 2020-06-05:
@epoch1970 recently did some work on the wireless AP documentation, so giving him a ping to see if this applies to his changes...?

burtyb, 2020-06-05:
I just put any bridge/static/etc. interface config in /etc/network/interfaces.d/somefile (which isn't checked before starting dhcpcd) and add the interface(s) to a "denyinterfaces ..." line in /etc/dhcpcd.conf . I'd imagine similar would work here too?

epoch1970, 2020-06-05:
I’m afraid I don’t understand. Defining a static IP config in dhcpcd is possible, is it not?

aallan, 2020-06-05:
Yes, e.g. https://www.raspberrypi.org/forums/viewtopic.php?t=140252

Lewiscowles1986, 2020-06-07:
Looks like the above is 50% of 

> I would also like to know if there is a way to provision static interfaces via the dhcp and see that linked in docs.

epoch1970, 2020-06-07:
https://www.raspberrypi.org/documentation/configuration/tcpip/README.md
????

Lewiscowles1986, 2020-06-07:
https://github.com/raspberrypi/documentation/commits/master/configuration/tcpip

epoch1970, 2020-06-07:
So do you agree the ticket can be closed?

Lewiscowles1986, 2020-06-07:
sure. The guidance is clear and I'll be updating my script(s) to use the one file.