raspberrypi/documentation #728: SD Formatter Tool no longer has "FORMAT SIZE ADJUSTMENT" option

woter324, 2017-09-17:
As per above. Scanned documentation and I cannot see a reference to it.

(Windows 7)

JamesH65, 2017-09-19:
Do you have a link to the docs page where this is a problem?

XECDesign, 2017-09-22:
This looks like the only occurrence:
https://github.com/raspberrypi/documentation/blob/master/installation/sdxc_formatting.md

paul-hammant, 2017-10-08:
https://www.raspberrypi.org/documentation/installation/noobs.md is has out of date advice. Find FAT in page.

Lukas-J, 2017-10-08:
The new tool make the format size adjustment automatically, I think.

paul-hammant, 2017-10-09:
'The new tool'  ?  URL please..

Lukas-J, 2017-10-09:
https://www.sdcard.org/downloads/formatter_4/index.html

hikarus3j, 2018-01-05:
Hello, I have a same question so I ask to helpdesk@sdcard.org about this.
And they send those message to me.

> The SD Formatter does not give an option to select the File System (FAT32 or exFAT): 
> 
> cards 64GB and above, the file system is automatically taken as exFAT
> cards 32GB and below, the file system is automatically taken as FAT32

can this help anyone?

paul-hammant, 2018-01-05:
Which page do you think those three lines should be added to?  URL pls :)

hikarus3j, 2018-01-09:
If I don't mistake your words, May be, that should be user Manuals, bottom of this page, I think.
https://www.sdcard.org/jp/downloads/formatter_4/

paul-hammant, 2018-01-09:
No, I'm asking which raspberrypi page do you think those three lines should be added to, to aid newbies ?

hikarus3j, 2018-01-10:
Oh... OK I understand, sorry.
but, I don't know about documents for raspberry pi newbies, I'm newbie too.
There is any document page about it ? I have no idea.
I'm afraid to say this, but does anyone answer paul's question?
please write it, I hope.

JamesH65, 2018-02-19:
I have fixed up the docs to remove the size adjustment reference.

closed 2018-02-19 by JamesH65