raspberrypi/documentation #729: Wireless Interface disapear after config via terminal

BORNintel, 2017-09-18:
Hi
I followed the instructions on this page:
https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/wireless-cli.md
after which i rebooted and my wireless interface was gone.

How do i fix this?

JamesH65, 2017-09-19:
This is a github for documentation issues, not technical support which is done on the Raspberry Pi forums. This is probably a Stretch issue, so I suspect the docs may be slightly out of date, I'll check them when time allows.

XECDesign, 2017-09-19:
That particular doc looks good - wpa_supplicant.conf information is still valid. There's nothing there that would make the interface disappear entirely.

JamesH65, 2017-09-20:
@BORNintel Please repost on the forums which are a better place for technical support questions.

closed 2017-09-20 by JamesH65