raspberrypi/documentation #741: Application OMXPlayer Is Missing

mobluse, 2017-10-09:
On https://github.com/raspberrypi/documentation/blob/master/raspbian/README.md
the Application OMXPlayer is missing.

bennuttall, 2017-10-09:
That's covered in [usage/video](https://github.com/raspberrypi/documentation/blob/master/usage/video/README.md)

mobluse, 2017-10-09:
I mean, it's missing among the links in the page I mentioned. Now only Camera is under Applications, but OMXPlayer should be there also. The page you just mentioned is a simplified version of https://github.com/raspberrypi/documentation/blob/master/raspbian/applications/omxplayer.md

If you click Applications you get to https://github.com/raspberrypi/documentation/blob/master/raspbian/applications/README.md
but Camera has a direct link from https://github.com/raspberrypi/documentation/blob/master/raspbian/README.md

bennuttall, 2017-10-09:
Oh, I see. I didn't know that page was there. Thanks!

JamesH65, 2018-02-19:
This appears to have been fixed.

closed 2018-02-19 by JamesH65