raspberrypi/documentation #745: Documentation NEEDS updates

svin83, 2017-10-18:
SETTING WIFI UP VIA THE COMMAND LINE

- https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

- https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/wireless-cli.md

This does NOT work with on a newly written SD card with Raspbian Stretch.
Documentation on the official page, that isn't working is worthless, and it forces people to google around, and the only info that shows up are outdated tutorials.

Stop fixing stuff that isn't broken, or make sure the documentation is up to date before wrecking everything like this...

NONE of the older tutorials for setting up network bridges, access ponts, proxies, tor relays etc are working now, and there's NO WAY for most people to get their head around problems without documentation...

JamesH65, 2017-10-18:
I have always found that being polite when asking for changes is a much better approach than going all ranty. Just a suggestion.

I am aware that there are now documentation issues in this area, after the Stretch upgrade. We have one PR on the docs repo about AP's that needs technical checking before the docs can be changed, and I am sure there are other areas. Unfortunately, I am very busy at the moment, and have been unable to spend the time on the docs that they really need. 

Interesting to note, that if we had changed all the documentation to match Stretch prior to releasing Stretch, Stretch would not yet have been released.  With limited manpower, something has to give. In this case, it the documentation will lagging behind the software. This always happens.

svin83, 2017-10-18:
Sorry if I come off as rude or abrasive, but I was really annoyed... There's NO info to find on how to set a static IP, or getting hostapd or dnsmasq working in stretch...

Trying to do it the old way will ruin connectivity... the only solution to these problems right now is to download older distros...

And it would be a good idea to make sure the outdated articles are marked as such... So people doesn't have to spend many hours trying and failing, and reflashing sd-cards many times...
trying to get things working with only misinformation available on the officisl site...

JamesH65, 2017-10-18:
The reason we have this documentation in github like this is so that third parties can edit it and make corrections, offloading the work from people who have more technical work to do (bug fixes, new products etc). Unfortunately, we get very few contributions.

svin83, 2017-10-18:
I'm sorry to hear that. I could probably contribute some minutes now and then, but I have no idea how to set stuff up in Stretch, so there's not much I could do.

XECDesign, 2017-10-18:
I should probably clear up what's different about stretch and why the changes were made.

It was required for the network to work out of the box on Stretch and increases compatibility. Some wireless devices come up at eth* rather than wlan* and with predictable network interface names, they would have to be determined from the MAC address. By relying on dhcpcd's own wpa hook instead of the interfaces file to launch wpa_supplicant, the configuration becomes simpler.

The original Debian package for dhcpcd would refuse to run if the interfaces file is configured. We lost that check when we added a systemd service for it, but added it back in Stretch. The problem is that people were setting static IP in the interfaces file and then wondering why that was getting changes by dhcpcd. The two systems are not meant to co-exist, so now if you need to decide whether you want to use ifupdown or dhcpcd. 'manual' interfaces are still supported in ifupdown, since it can be used to disable wifi power management, for example.

When you open the interfaces file, there is a note at the top explaining that you should use dhcpcd to configure the system. dhcpcd.conf has some static IP examples. All the information you need can be found in dhcpcd, dnsmasq or hostapd man pages. I don't think it's fair to say that there is NO info.

None of these changes affect the page you linked. That page is still 100% valid as is. One problem is that people misinterpret 'add' as 'replace'. Instead of appending to the wpa_supplicant.conf file, some people have been replacing its content with just a network block. This used to 'work', but no longer does. The existing information in that file allows users in the netdev group to use tools like wpa_cli to control wpa_supplicant. This is functionality that dhcpcd relies on and is a part of correct setup for wpa_supplicant in general.

If you're having trouble getting something to work, I think it's best to post on the forum first to figure out why exactly.

svin83, 2017-10-18:
I tried to use the info in that page to get my headless pi connected to wifi. I tried, failed, googled, tried, and still failed...

The only way I could get my pi3 to connect to my wifi was by adding a wpa_supplicant.conf file to /boot before first boot... I still couldn't find my wifi settings anywhere, so I have no idea how to connect manually...

The info in the page I linked did not help me, so I assumed that it is the same for every pi3 on stretch since 3 different boards (all pi3) refused here...

XECDesign, 2017-10-18:
For headless setup, all you need to do is put wpa_supplicant.conf and an empty 'ssh' file on the boot partition. wpa_supplicant.conf gets moved to /etc/wpa_supplicant/wpa_supplicant.conf.

If this doesn't work, please make a forum post with the content of your wpa_supplicant.conf file, what you expect to happen, what actually happens, the output of 'sudo journalctl' , if possible, and how the card was imaged. We can then figure out why it's not working for you and make appropriate changes to the documentation if necessary.

lurch, 2017-10-18:
> I could probably contribute some minutes now and then, but I have no idea how to set stuff up in Stretch, so there's not much I could do.

**Politely** pointing out which instructions don't work, and opening a separate issue for each (with a clear explanation of what does and doesn't work, any error-messages you see, etc. etc.), allows those people who _do_ have time to work on the docs (which unfortunately isn't me either), to know which parts need fixing ;-)

svin83, 2017-10-18:
Well... That's why i mentioned it... I found it in the forum somewhere while googling around...

Putting it in /boot works, and it should be in the documentation under "connect to wireless from cli".

Editing the /etc/wpa_supplicant/wpa_supplicant.conf directly does NOT work.

So my guess is that the init script does something more than just expanding the FS and copying the wpa_supplucant.conf file to /etc/wpa_supplicant/.

I have no idea where I can see what it does, so manually changing wifi later is an excercise in tedium with the absence of updated docs...

svin83, 2017-10-18:
If I knew all that @lurch, I would...

XECDesign, 2017-10-18:
No, that's really all it does - move the file:
https://github.com/RPi-Distro/raspberrypi-net-mods/blob/master/debian/raspberrypi-net-mods.service#L9

macfree, 2017-10-18:
Hey, just want to add that it worked on rpi3b with raspbian stretch lite  version 2017-09-07. But after a "sudo reboot". Should it be added to the instruction ?

XECDesign, 2017-10-18:
@macfree did you run `wpa_cli reconfigure`?

XECDesign, 2017-10-19:
Just ran through the documentation to check why a reboot was required.

It should be `wpa-cli -i wlan0 reconfigure`, because wpa_cli defaults to `p2p-dev-wlan0` instead of `wlan0`. Sent an appropriate PR for this.

macfree, 2017-10-19:
@XECDesign yes did ran reconfigure (but not with the -i flag).

PR #746 (2017-10-23): Update wireless-cli for stretch
Fixes #745
closed 2017-10-23 by JamesH65