raspberrypi/documentation #749: please clarify location for ssh file as /boot

quixand, 2017-10-22:
i've just spent 30 minutes of my life trying to enable ssh using the ssh file, because i was putting it in / rather than /boot

I can't be the only person making this mistake (I hope im not!)
thanks

quixand, 2017-10-22:
ive made the changes that would have helped me in a fork here https://github.com/quixand/documentation/commit/b730253d539c1b0325bb8aa7529cbb9958e7c609 I can submit a pull request but which branch would you like me to use?

XECDesign, 2017-10-23:
If you say /boot, people will put it either in the /boot directory of the boot partition or the /boot directory of the root partition and neither is the right location.

The instructions right now are correct, but it may be worth clarifying terminology and explaining how a directory differs from a partition and what a mountpoint is.

XECDesign, 2017-10-23:
Also, how would you run `touch /boot/ssh` on a headless pi without ssh enabled already?

quixand, 2017-10-23:
I think you are confusing a booted pi from using the sd card in another machine. I just booted the pi with a monitor and keyboard and did "sudo touch /boot/ssh"

It may not be clear to others that the root partition is not the boot partition.

XECDesign, 2017-10-23:
That's not quite headless setup then is it? Normally you'd enable ssh using `systemctl enable ssh` or `raspi-config`. The boot partition mechanism is there specifically for headless setup.

I am all for clarifying how it works and where the file needs to go. I'd just prefer to do it in a way that doesn't conflate mountpoints, partitions and directories, which would only add ambiguity to where the file should actually go.

quixand, 2017-10-23:
ok, I have corrected my branch based on your comments.
https://github.com/quixand/documentation/commit/e841a23e8b6cd6db79873f5243995024490efd72#diff-3022880829cc73a6f254762d548329ec

XECDesign, 2017-10-23:
Looks good

JamesH65, 2017-10-24:
Can you submit this as a PR please, on the master branch.

quixand, 2017-10-24:
done

closed 2017-11-20 by JamesH65