raspberrypi/documentation #761: Suggesting that add the method of how to install Caddy on Raspberry Pi

reechang, 2017-11-07:
Caddy is is an open source, HTTP/2-enabled web server written in Go.
It's very light , and easy to dispose.
But the information about how to install it in the Raspberry is very limiting.

bennuttall, 2017-11-07:
Are you offering to write it or asking us to write it? If you can write it, or have a tutorial you can link to, we'll consider adding it. But I can't see anyone here finding the time to do this - we tend to cover majority use case documentation. We wrote the Apache page, and accepted a PR adding the nginx page.

JamesH65, 2017-11-10:
As Ben implies, we do not have time to add entirely new documentation pages for relatively low use cases, but if you write some docs  we do have time to technical check and copy edit them.

JamesH65, 2017-11-20:
No response, closing. If you wish to write a PR for this, we are happy to look at it for inclusion.

closed 2017-11-20 by JamesH65