raspberrypi/documentation #762: An access point to share an internet connection

peylight, 2017-11-09:
Hello
I have some problem with [this](https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md#using-the-raspberry-pi-as-an-access-point-to-share-an-internet-connection) document.
I try to do that but when i run this command:
`$ sudo brctl addbr br0`
The output is like that:
`add bridge failed: Package not installed`

texadactyl, 2017-11-09:
Cannot reproduce your issue in Raspbian stretch.
uname -a
Linux raspberrypi 4.9.60-v7+ #1048 SMP Fri Nov 3 16:05:21 GMT 2017 armv7l GNU/Linux

Fix through upgrade?

peylight, 2017-11-09:
```
$ uname -a
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

```
I found the solution!
Need reboot to load module into kernel.

texadactyl, 2017-11-09:
Should not need to reboot. I installed hostapd bridge-utils and it worked right away (the kernel module was added dynamically).  If this is not the case in your Raspberry Pi Linux distro (?), then
```
sudo modprobe {with the full path of the kernel module}
```
should also work without the need for reboot.

JamesH65, 2017-11-20:
There is an pending PR to fix this page for use with Stretch.

closed 2017-11-20 by JamesH65