raspberrypi/documentation #765: in man no documentation - how to properly configure the wifi.

Packuh, 2017-11-13:
[](https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=197378&p=1233501#p1233501)

Tell me pls how to properly describe these lines, if you can not change the file /etc/network/interfaces?

file /etc/network/interfaces

```
allow-hotplug wlan0
iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```

I added these lines with my hands and after that everything worked. I first encountered such an innovation. Earlier i used /etc/network/interfaces and everything was fine. without these lines the wifi does not work. Where is it prescribed?

But I do not understand why in the / etc / network / interfaces it is written that it is not necessary to edit it.

```
# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'
```

XECDesign, 2017-11-13:
There is no need to edit `/etc/network/interfaces`. If you're having problems, could you please post your `/etc/wpa_supplicant/wpa_supplicant.conf` and the output of `sudo journalctl` to that forum thread.

JamesH65, 2017-11-13:
This is a question for the forum, so closing.

closed 2017-11-13 by JamesH65

texadactyl, 2017-11-15:
Never edit /etc/network/interfaces.  Doing so negates the use of the GUI Network Manager which runs on desktops.

You might find this how-to useful: [(https://www.element14.com/community/community/raspberry-pi/blog/2016/09/16/how-to-set-up-a-static-fixed-ip-address-on-raspberry-pi)]