raspberrypi/documentation #767: DHCP breaks eth0 on Pi3B

h-schwanbeck, 2017-11-16:
Hello,

I've got two Raspberrry Pi 3 Model B on my desk with the same strange behaviour.
Sometimes when sending data via eth0 it just stops working and ifconfig shows TX errors.
Once this bug has been triggered I can get eth0 to work for another minute by bringing the interface down and back up again. Then it reoccurs on its own.

Steps to reproduce:
1) Copy Raspbian Stretch (Lite); Version: September 2017 to SD card
2) Boot Raspberry Pi from SD card; eth0 connected to DHCP enabled network
3) Login and run "sudo systemctl start ssh"
4) Login via SSH from a different machine
5) Run "sudo raspi-config"

After that I only see a blue terminal on the remote machine. The Raspi keeps running, but eth0 does not work anymore. ifconfig shows increasing "TX Errors".
I tried two Raspis, different cables and switches - same behaviour.
Also "rpi-update" did not help.

But, the bug did not occur after I changed the dhcpcd.conf to use static IP addresses.
So I guess this is a software problem somehow tied to DHCP.

Best regards,

Henry Schwanbeck

JamesH65, 2017-11-16:
Any error messages in syslog?

h-schwanbeck, 2017-11-16:
Hi James,
I saw nothing, not in syslog, messages nor in dmesg

JamesH65, 2017-11-16:
In step 5, does that need to be raspi-config, or does running anything cause issues?

h-schwanbeck, 2017-11-16:
No it's not anything, I chose raspi-config because it reproduced every single time.
But sometimes it happened when I opened a file using vi or nano.
May be these programs send a command sequence that triggers the bug.
I have to say that I only tested SSH, no web server or anything.

JamesH65, 2017-11-20:
Might be worth posting this on the forums, much larger audience. This issue tracker is for problems with the documentation.

JamesH65, 2018-02-19:
Closing, not a documentation issue.

closed 2018-02-19 by JamesH65