raspberrypi/documentation #773: Imaging with Etcher does not work when using mSata to USB Converter

c64emulator, 2017-11-28:
The Documentation recomments a tools named Etcher for imaging USB mass storage device.
This tool does not work when using a mSata device which is connected via mSata to USB Converter (like piDRIVE from Kickstarter or the Lycom L102).
Please mention alternative software for this cases.
TNX.

JamesH65, 2017-11-28:
Does Win34diskImager work correctly on that device? Might also be worth reporting this as a bug to the Etcher team - this is the only time I've heard of it not working.

lurch, 2018-02-11:
@c64emulator It might be that Etcher is detecting your mSata to USB converter as a non-removable drive? If you go into Etcher's settings (cog icon in the top-right) and enable "Unsafe mode", does the drive then appear in Etcher?

JamesH65, 2018-02-19:
@c64emulator Did @lurch comment help with this? We have not had any other reports on Etcher not working, so I'm included to think this may be a one off.

JamesH65, 2018-03-12:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2018-03-12 by JamesH65

tsn77130, 2018-03-28:
@JamesH65 yes, unsafe mode works, thanks !

c64emulator, 2018-04-05:
In the last week I wrote a image to a new mSATA-SSD with the most current version of Etcher. It seems to work for me now. The only thing I have now: the other way round (making a backup image) does not work with Etcher. Therefore the Win32 Disk Imager is the tool of choice for me.