raspberrypi/documentation #775: driver=nl80211 line causes error on Pi Zero W and does not work

steviebald, 2017-11-29:
I tried following the instructions on a Pi Zero W and could not get it to work.  A Linux sysadmin at work helped me track down the issue to the driver=nl80211 line as there was an error saying driver not found.  When we removed that line it worked perfectly.

JamesH65, 2017-11-29:
Can you indicate which instructions you were using that had this error?

steviebald, 2017-11-29:
https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md

for setting up pi as a standalone access point

JamesH65, 2017-11-29:
In the final section there is the following...

> The access point setup is almost the same as that shown in the previous section. Follow the instructions above to set up the hostapd.conf file, but add bridge=br0 below the interface=wlan0 line, and remove or comment out the driver line.
> 
> interface=wlan0
> bridge=br0
> #driver=nl80211
> ...

steviebald, 2017-11-29:
I stopped at the end of the first section as I wasn't trying to set it up to share internet access, just as an access point so I can connect to and control a robot.

JamesH65, 2017-12-01:
Odd. When I wrote that page, I needed to have that line in there. However, we have moved on to Stretch, which has changed quite a lot of things that need to be updated in that document. Perhaps this is one of them.

steviebald, 2017-12-01:
I was trying it on a Pi Zero W, not sure if that is any different.  Apart from that, the instructions were great, keep up the good work!

aallan, 2020-06-03:
Open since 2017. Still relevant? If not can we close?

JamesH65, 2020-06-03:
That page has been completely changed now, so closing.

closed 2020-06-03 by JamesH65