raspberrypi/documentation #779: security.md: Use adduser instead of useradd

Garcia98, 2017-12-03:
[Link to the page to which the issue relates - in "Changing your username section"](https://www.raspberrypi.org/documentation/configuration/security.md)

The command "useradd" is deprecated and "adduser" should be used whenever it's possible.

The use of useradd brings undesirable consequences like sh being the login shell instead of bash, these problems are avoided if adduser is used instead.

From adduser manual page:
>        They  [adduser and addgroup] are  friendlier  front ends to the low level
>        tools like useradd, groupadd and usermod programs, by default  choosing
>        Debian  policy conformant UID and GID values, creating a home directory
>        with skeletal configuration, running a custom script,  and  other  fea-
>        tures.

From useradd manual page:
>        useradd is a low level utility for adding users. On Debian,
>        administrators should usually use adduser(8) instead.

[Link to the proposed change](https://github.com/raspberrypi/documentation/commit/3814dfeb83ce55e856729315ee81857f66cb2530)

bennuttall, 2017-12-05:
Agreed - I wrote the guide on [adding users](https://www.raspberrypi.org/documentation/linux/usage/users.md) which uses `adduser` and this guide (if necessary?) should do too.

JamesH65, 2017-12-05:
@bennuttall Are you going to do a PR?

bennuttall, 2017-12-05:
Can do!

JamesH65, 2018-03-05:
ping @bennuttall

JamesH65, 2018-03-21:
@bennuttall  Just had this reported again, so probably need to prioritise this one.

MaxVMH, 2018-03-21:
@JamesH65  I proposed a fix (fix #862).

Sorry if this is not sufficient or how we are supposed to propose changes, but thanks for giving this issue some attention anyway.

PR #862 (2018-03-21): Changed 'useradd' command to 'adduser'
Proposed change to fix #779 
I removed the instructions to change the password since 'adduser' asks to enter a password by itself.
commit f56ef186 (2018-03-21): fixes #779

Proposed change to fix #779
closed 2018-03-21 by bennuttall