raspberrypi/documentation #791: wpa_supplicant.conf

ribbox, 2017-12-30:
The instruction to create a wpa_supplicant need to be updated to reflect changes in Stretch. 

The following need to be added to the start of the file;

```
country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```

Where country code relates to the country you use the RPi in. #790

JamesH65, 2018-03-05:
Sorry about the delay. Can you post a link to the specific page that you think needs correcting?

philwilkinson40, 2018-03-30:
https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

JamesH65, 2018-07-03:
@philwilkinson40 I made some changes to specify the country code setting on the page you linked - do you think it's now OK or is more work required?

JamesH65, 2018-09-12:
Closing this issue as questions answered/issue resolved.

closed 2018-09-12 by JamesH65