raspberrypi/documentation #793: bootflow.md: please clarify how to disable boot modes via GPIO

rd235, 2018-01-02:
bootflow.md says:
*Uses program_gpio_bootmode to disable some modes by reading GPIOs 22-26 or 39-43 to see if the default values do not equal the default pull to '0'. If it is low, it will disable that boot mode for each of SD1, SD2, NAND, SPI, USB. If the value read is a '1', then that boot mode is enabled (note this cannot enable boot modes that have not already been enabled in the OTP). The default pull resistance is around 50K ohm, so a smaller pull up of 5K should suffice to enable the boot mode but still allow the GPIO to be operational without consuming too much power.*
This seems to mean that all boot mode are *disabled* by default: in fact, all the pins 22-26 or 39-41 are down at startup time.
(see pag 102 of https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf -- these values should be the same for BCM2837)
So, from your boot flow description at least one 5K resistor should be needed to *enable* the desired boot mode.
On the contrary there are on the net several videos showing how to boot a RPi3 from MSD or network, and no one mentions about the need of resistors.
Please rephrase the sentence above to clarify when pull up resistors are needed and which is the behavior with and without the pullup.
Thank you.

JamesH65, 2018-02-19:
Ping @ghollingworth Is this one for you?

ghollingworth, 2018-02-19:
The above is only valid when program_gpio_bootmode is set...  Once this bit has been set then the boot modes will be disabled, you'll then require a pull up to re-enable your specific boot mode

Gordon

JamesH65, 2018-02-19:
Do we need to clarify the docs?

ghollingworth, 2018-02-20:
No I don't think so, it says this at the top of the document.  Although James, maybe just have a read through and make sure you agree...  Obviously being the person to write the data does tend to give one a different viewpoint!

rd235, 2018-05-14:
There are two OTP flags: _program_usb_boot_mode_ and _program_gpio_bootmode_. Somewhere I would say that _program_gpio_bootmode_ is *not* needed to boot a RPI from a MSD or from the network. (am i misundertanding the specifications?)
When program_gpio_bootmode is UNSET, are all the devices enabled (SD1, SD2, NAND, SPI, USB) or just SD1 and USB (provided _program_usb_boot_mode_ is 1)?

ghollingworth, 2018-05-15:
There is a bootmode register enabling the other boot modes, program_gpio_bootmode cannot enable bootmodes, only disable them.

So you've only got SD1 and USB enabled

andrum99, 2019-06-11:
> There are two OTP flags: _program_usb_boot_mode_ and _program_gpio_bootmode_. Somewhere I would say that _program_gpio_bootmode_ is _not_ needed to boot a RPI from a MSD or from the network.
Correct. GPIO boot mode is a special boot mode which checks certain GPIO pins to determine at boot time which boot modes should be attempted. If all you want to do is use MSD or net boot, then you don't need GPIO boot mode. program_usb_bootmode is the one you need to enable both MSD and ethernet boot, despite the name.

andrum99, 2019-06-11:
> There is a bootmode register enabling the other boot modes, program_gpio_bootmode cannot enable bootmodes, only disable them.
> 
> So you've only got SD1 and USB enabled

Most of the docs refer to SD0 and SD1 rather than SD1 and SD2. To be clear, it is the "first" SD/MMC interface that is used by default on the Raspberry Pi. This is the Broadcom SD/MMC interface. The second SD/MMC interface is from Arasan and is also capable of SDIO, hence it is used to talk to the Wifi chip on the various models of Pi which have built-in wifi.

lurch, 2019-07-03:
Fixed by #1131

closed 2019-07-03 by lurch