raspberrypi/documentation #801: Idea: Glossary

madneon, 2018-01-14:
It would be useful, to provide **Glossary** section in Documentation.

There is many abbreviations used in text, and the only way to "decipher" them is to search the web, Wikipedia etc. - especially for new users trying to learn Raspberry world.

JamesH65, 2018-01-14:
Nice idea. If anyone interested in contributing can add a comment to this issue, I'll collate them all and create a page.

madneon, 2018-01-14:
Those terms I found across pages of Documentation with no explanation or links:

- GPIO
- SPI
- I2C
- SoC
- UART
- DT (Device Tree), DTS (Device Tree Source), DTB (Device Tree Blob)
- DPI
- SPI
- OTP
- NAND
- MSD
- TP1/TP2

Also maybe to worth to put into Glossary:
- config.txt
- raspi-config
- vcgencmd

Or even maybe some "non-Pi" terms, like:
- LED
- A/mA (amper/mili amper)
- V (Volts)
- DHCP

JamesH65, 2018-02-19:
Anyone have any ideas on the best place this could fit in the documentation hierachy? @bennuttall

bennuttall, 2018-02-19:
This is a great idea. I'd also like to add a documentation index tree to help people find pages without having to traverse it themselves.

Maybe the comms team can help with the glossary of terms.

I think both of these could live in files in the root of the repo.

bennuttall, 2018-02-19:
I'm making a start on this now and will push to a branch

JamesH65, 2018-02-19:
Thanks, I was just about to start on it too...but you can do it!

bennuttall, 2018-02-19:
> For the docs index tree, is it possible to genreat that automatically?

That's what I had in mind

> Woudl it also be possible to have metadata per page to act as a brief content descriptor that could be used in the index?

Probably not. I think it'll be possible to use the h1 from each page rather than the directory name, but 

I could either do it in PHP, and it'll purely live on the web server (not on GitHub), but will be generated and updated automatically, or I could do it in Python (probably easier) and the output can live on GitHub too - but that'll need re-running to keep it updated as the index changes (not very often tbh). Manual re-run wouldn't be EOTW but post-commit script would require some setup. I'll have a think. First: glossary.

bennuttall, 2018-02-19:
Pushed to https://github.com/raspberrypi/documentation/blob/glossary-801/glossary.md

JamesH65, 2018-03-01:
I've been adding stuff in as I think of it - any suggestions on where to put it in the docs tree.

JamesH65, 2018-03-12:
@bennuttall I think this is in a state where it can be linked in to the website (after copy edit). Any news of having it linked in the page footers?

bennuttall, 2018-03-12:
Not touched it yet but should be a fairly quick job. Just got a few things to deal with first and I'll get on it.

JamesH65, 2018-07-02:
@bennuttall Ping - I think this can now go live.

JamesH65, 2018-07-02:
@LizUpton Liz, any comments? Thinks its worth getting this up - we can add/adjust anything later. It's all been copy edited.

JamesH65, 2018-09-24:
Reopened this as I don't believe there is a link to the glossary on the website or in the docs, so it's a bit hidden. Any suggestions on where to link it from? I'd like it in the footer, or from the main page (with the 8 icons)

JamesH65, 2018-09-24:
Current version is here https://github.com/raspberrypi/documentation/blob/master/glossary.md

bennuttall, 2018-09-24:
I've renamed the file so it's at /glossary/ not /glossary.md

https://www.raspberrypi.org/documentation/glossary/

I will add a link to the docs homepage

lurch, 2019-06-11:
@bennuttall @JamesH65 Is the glossary actually linked to from anywhere yet? :confused:

bennuttall, 2019-06-11:
Doesn't look like it. We could add to the footer or just the homepage...

commit 2b52cf4b (2018-07-11): WIP: add glossary, close #801 (#864)

* Initial commit of glossary page

* Update glossary.md

A few additions.

* Update glossary.md

Added a few more items

* Update glossary.md

* Update glossary.md

* Add dt-blob.bin and Overlay (#841)

* Update glossary.md

* Update glossary.md

* Update glossary.md (#844)

* copy edits

* copy edits

* copy edits

* copy edits