raspberrypi/documentation #805: TFTP boot: warn users about network hardware incompatibility

jvsalo, 2018-01-15:
I believe the tutorial page https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md deserves a mention about the incompatibility of networking hardware.

I know it's mentioned here https://www.raspberrypi.org/blog/pi-3-booting-part-ii-ethernet-all-the-awesome/ but to save other peoples' time I would certainly mention it on the main tutorial page.

It can be frustrating to debug why nothing works, if the R-Pi firmware has trouble negotiating link with the switch/computer on the other end.

Most of my networking hardware was incompatible, finally got hold of HP 1810-24G that worked.

jvsalo, 2018-01-15:
Disabling STP frames in the network worked wonders, by the way.

JamesH65, 2018-01-16:
@ghollingworth Any thoughts Gordon?

JamesH65, 2018-02-19:
ping @ghollingworth

ghollingworth, 2018-02-19:
I do not have suitable hardware to test this on, so would only be adding, YMMV

If you have some specific examples of switches that do not work can you add a note as a pull request?

Thanks

JamesH65, 2018-09-12:
fixed by #983