raspberrypi/documentation #806: Possible change of tool used for SD card discovery in installation docs

MrCelticFox, 2018-01-16:
In the section on installing OSes onto SD cards from Linux [here](https://github.com/raspberrypi/documentation/blob/master/installation/installing-images/linux.md) the tool `df` is used to find the name of the SD card, however as mentioned if the volume isn't mounted then it won't show up and another command must be used to find it. What about using `lsblk`? I think this is a cleaner approach and it works whether the device is mounted or not (and in fact you can see if it is and where), it's also part of the `util-linux` group, so should be present be default in the user's distro.
Just wanted to get feedback before I did a pull request to find out if there are any issues people can see with using this?

Sample output from `df -h`:
```
Filesystem      Size  Used Avail Use% Mounted on
dev             7.8G     0  7.8G   0% /dev
run             7.8G  1.2M  7.8G   1% /run
/dev/nvme0n1p3  226G   18G  197G   9% /
tmpfs           7.8G     0  7.8G   0% /sys/fs/cgroup
tmpfs           7.8G  3.0M  7.8G   1% /tmp
/dev/nvme0n1p1  256M   39M  218M  16% /boot
tmpfs           1.6G   48K  1.6G   1% /run/user/1000
/dev/sda1       458G  218G  217G  51% /run/media/username/4238abc6-1234-5678-ffff-2934fcab8a0c
```

Sample output from `lsblk`:
```
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 465.8G  0 disk 
└─sda1        8:1    0 465.8G  0 part /run/media/username/4238abc6-1234-5678-ffff-2934fcab8a0c
mmcblk0     179:0    0  14.9G  0 disk 
└─mmcblk0p1 179:1    0  14.9G  0 part 
nvme0n1     259:0    0 238.5G  0 disk 
├─nvme0n1p1 259:1    0   260M  0 part /boot
├─nvme0n1p2 259:2    0     8G  0 part [SWAP]
└─nvme0n1p3 259:3    0 230.2G  0 part /
```
with `lsblk` the user can also clearly see the difference between disk name and partition name.

I have already forked the file in question and made changes, and will submit a pull request if the maintainers approve.

JamesH65, 2018-01-16:
Sounds like a good idea to me. Feel free to fire off a PR. Note, we are busy at the moment, might take a while for it to be checked and merged.

MrCelticFox, 2018-01-16:
Thanks for the quick response, when you have time see #807

lurch, 2018-02-03:
Yeah, https://github.com/raspberrypi/documentation/blob/master/hardware/computemodule/cm-emmc-flashing.md#writing-to-the-emmc---linux also suggests using `lsblk` :slightly_smiling_face:

JamesH65, 2018-02-19:
Fixed

closed 2018-02-19 by JamesH65