raspberrypi/documentation #811: Documentation links are broken

ayghor, 2018-01-29:
on

https://www.raspberrypi.org/documentation/

teh links

* https://www.raspberrypi.org/documentation/remote-access/
* https://www.raspberrypi.org/documentation/linux/
* https://www.raspberrypi.org/documentation/raspbian/

are broken and redirect to:

ayghor, 2018-01-29:
its also worth noticing that teh site is responding with 302 (found), redirecting to /404, which in turn responds with 404 (not found).

JamesH65, 2018-01-29:
Which page(s) are these links on?

ayghor, 2018-01-29:
@JamesH65 https://www.raspberrypi.org/documentation/

JamesH65, 2018-01-29:
@bennuttall @dtt101 Has anything changed recently that may have broken these links?

bennuttall, 2018-01-29:
Not that I'm aware of - looking in to it...

bennuttall, 2018-01-29:
Resolved!

It looks like the download from github only picked up half of the contents. I've restored it now.

I'll look into why it might have happened but it's surely just a blip.

bennuttall, 2018-01-29:
Here's the log entry:

```
Mon Jan 29 11:15:01 GMT 2018 - Updating documentation
tar: Substituting `.' for empty member name
tar: .: Cannot open: Is a directory
tar: Substituting `.' for empty member name

gzip: stdin: invalid compressed data--format violated
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
Mon Jan 29 12:15:01 GMT 2018 - Updating documentation
tar: Skipping to next header
tar: Exiting with failure status due to previous errors
```

closed 2018-01-29 by bennuttall

JamesH65, 2018-01-29:
Thanks @bennuttall

dtt101, 2018-01-29:
Thanks @bennuttall !

bennuttall, 2018-01-29:
~Thanks @bennuttall~

Thanks @ayghor for reporting it :+1:

ayghor, 2018-01-30:
<3