raspberrypi/documentation #822: Clearer documentation for installing operating system images on Linux

wisdomlight, 2018-02-10:
From my point of view - a beginner the following code to check if an image was correctly written to a SD card - is not very clear.

`dd bs=4M if=/dev/sdX of=from-sd-card.img
truncate --reference 2017-11-29-raspbian-stretch.img from-sd-card.img
diff -s from-sd-card.img 2017-11-29-raspbian-stretch.img`

1. the meaning of `from-sd-card.img` is not clear and actually for a beginner quiet confusing. 
I assume the words from-sd-card.img refer to the .img that was just written to the SD card. But what is not clear is do i need to state `of=/dev/sdb/xxxx.img`  ??? or just is `rom-sd-card.img` [it may be obvious to you that it is just a place holder for the actual comman, however for a beginner this is not clear]
this point is applicable to all the places where it says from-sd-card.img

2. the following 
`dd bs=4M if=/dev/sdb of=NextCloudPi_02-06-18.img truncate --reference NextCloudPi_02-06-2018.img /dev/sdb/NextCloudPi_02-06-2018 diff -s ~/Downloads/NextcloudPi_02-06-2018/NextCloudPi_06-02-2018.img  `
returned the following
`dd: invalid option -- '-'
Try 'dd --help' for more information.
`

so somewhere something does not work right

wisdomlight, 2018-02-11:
OK thanks
Actually this point you made

> but there's actually 3 separate commands there (each of which needs to be typed on a separate line) :

is a good point that sometimes not very clear in the raspberry pi instructions.

thanks for taking the time explaining to me.

And yes I do use etcher very often. I felt like trying to be a super hero using only the cli....;-)

cheers

JamesH65, 2018-03-02:
Proposed change here https://github.com/raspberrypi/documentation/pull/838

lurch, 2018-04-11:
Now that #838 has been merged, I guess this can be closed?

JamesH65, 2018-04-11:
Yup!

closed 2018-04-11 by JamesH65