raspberrypi/documentation #824: Use adduser instead of useradd in "Securing your Raspberry Pi"

Benestar, 2018-02-15:
When using `useradd` all the default options are apparently ignored. [1] This especially leads to the wrong shell being selected and also some other defaults missing. Therefore, it should rather be suggested to use the `adduser` command which does not have these problems. See also [2] for more information.

[1] From `man useradd`: "useradd is a low level utility for adding users. On Debian, administrators should usually use adduser(8) instead."
[2] https://unix.stackexchange.com/a/321235

JamesH65, 2018-02-19:
Woudl you be up for doing a PR to update the page? Seems like a good idea to me.

JamesH65, 2018-02-19:
Ah, just seen this is a duplicate of https://github.com/raspberrypi/documentation/issues/779

I'll close this and can be dealt with on other issue.

closed 2018-02-19 by JamesH65