raspberrypi/documentation #827: Add an index tree

bennuttall, 2018-02-19:
As suggested in #801, we should add an index tree containing the docs index.

From #801:

> For the docs index tree, is it possible to genreat that automatically?

That's what I had in mind

> Woudl it also be possible to have metadata per page to act as a brief content descriptor that could be used in the index?

Probably not. I think it'll be possible to use the h1 from each page rather than the directory name, but getting a meaningful description of the page would be tricky. I think the page title should be sufficient.

I could either do it in PHP, and it'll purely live on the web server (not on GitHub), but will be generated and updated automatically, or I could do it in Python (probably easier) and the output can live on GitHub too - but that'll need re-running to keep it updated as the index changes (not very often tbh). Manual re-run wouldn't be EOTW but post-commit script would require some setup. I'll have a think. First: glossary.

bennuttall, 2019-04-23:
@grega do you think your static site builder could add a site index made up of a tree of headings like my attempt here: https://github.com/raspberrypi/documentation/blob/827-index-tree/index.md

grega, 2019-04-23:
@bennuttall yep I reckon so! Will have a stab at it shortly...

aallan, 2021-06-07:
Superceded by #1911

closed 2021-06-07 by aallan