raspberrypi/documentation #833: Documentation for the "initramfs" option doesn't mention all options for the second parameter

AlexanderKurtz, 2018-02-22:
According to the current documentation [0], the ```initramfs``` should be used like this:

```
initramfs initramf.gz 0x00800000
```

However, apparently omitting the memory location [1] or using the special value ```followkernel``` [2] is also possible here. Could somebody with access to the source of The Blob maybe comment what happens in these cases and add this to the documentation?

[0] https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/boot.md#initramfs
[1] https://anonscm.debian.org/cgit/pkg-raspi/raspi3-firmware.git/tree/debian/kernel/postinst.d/raspi3-firmware#n90
[2] https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/conditional.md#the-pi1-and-pi2-etc-filters

JamesH65, 2018-03-01:
The follow kernel options is now documented here

https://github.com/raspberrypi/documentation/blob/master/configuration/config-txt/boot.md#initramfs

closed 2018-03-01 by JamesH65

AlexanderKurtz, 2018-03-02:
Thanks @JamesH65!

However, what about the case when the second argument is omitted? What does that mean/do?