raspberrypi/documentation #843: Set default shell for new users

sertal70, 2018-03-11:
The [Securing your Raspberry PI](https://github.com/raspberrypi/documentation/blob/master/configuration/security.md#securing-your-raspberry-pi) document, in "Change your username" paragraph  should specify that after having aded a new user it is required to set the default shell using the command:

`sudo usermod -s /bin/bash fred`

Failing to do so lead the new user to strange behaviors, like terminal windows showing only the menu bar and title bar.

XECDesign, 2018-03-11:
Is that true once https://github.com/raspberrypi/documentation/issues/779 is fixed?

sertal70, 2018-03-17:
No, actually when #779 will be fixed this issue is fixed, too. I'm going to close it, thanks!

closed 2018-03-17 by sertal70

XECDesign, 2018-03-17:
Thanks for checking.