raspberrypi/documentation #847: CMCDA v1.1 Schematic Incorrect

tegan-lamoureux, 2018-03-14:
Hi there,

Just a heads up, the pin numbering on [J4 of this schematic](https://github.com/raspberrypi/documentation/blob/master/hardware/computemodule/RPI-CMCDA-V1_1-SCHEMATIC.pdf) is backwards. It's also possible J2 is backwards for the same reason, although I haven't verified this. 

I was using it as a reference for a custom Compute Module 3 board, and realized when we hooked it up that I was shorting 3.3V to ground. The reversing is clear when visually inspecting it, and I also verified this with a multimeter. I have v1.1 of the board.

It's a pretty valuable schematic, as it's one of only two ways to find a pinout of the ribbon cable for the foundation's official 7" display (the second being the PI3 schematic, which uses a different connector). Publicly listing the ribbon cable pinout in the display's documentation would also be a massive help!

Thank you for your help! Image of connector: (Pin 15 is on the left.) 

[image]

JamesH65, 2018-03-19:
@Roger-Thornton any thoughts?

Roger-Thornton, 2018-03-19:
I believe @mike-stimson designed this board so I will have to defer to him for comment.

lurch, 2018-03-19:
I wonder if this might be anything to do with https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=80508 ?

mike-stimson, 2018-03-19:
This isn't my doing, but I think I can guess why this has happened. I think the intention has been to maintain the pin numbering w.r.t. Pi3, eg, DSI_D1_N is pin 2 on both the adaptor and Pi3. I'll admit this could be a lot clearer though.
These kinds of things are always going to be confusing on this type of connector where adaptors are involved. The Pi3 schematic should always be used as a reference for connector pinouts.
Incidentally, if the pinout has been accidentally reversed on a design, then a flex with the contacts on opposite sides of the flex (ie, top and bottom) can be used (or both on the same side, depending on where the inversion is).

tegan-lamoureux, 2018-03-19:
It's not really that confusing. It's just factually incorrect. 3.3v doesn't go to pin 14 and 15 of the SFW15R-2STE1LF connector as routed on the schematic, it goes to pin 1 and 2. The connector has pin 1 and 15 physically labeled in the moulded shell, and the signals don't match that numbering.

Anyways, just thought I'd bring it up. As you said, there's plenty of cables and top/bottom contact connectors that make this an easy fix. Thank you.

JamesH65, 2018-09-12:
@mike-stimson @tegan-lamoureux So does something need to change in the documentation?

Note, file in question has changed name and moved. 
https://www.raspberrypi.org/documentation/hardware/computemodule/schematics/rpi_SCH_CMCDA_1p1.pdf

tegan-lamoureux, 2018-09-12:
The schematic is incorrect, so I would say yes. Or at least make a note of it. Anyone with one of these adapters can confirm this.

JamesH65, 2018-10-01:
@mike-stimson Are we likely to be able to fix this in the nr future?

JamesH65, 2018-10-01:
Seems this is more @jimbojr issue. Any thoughts?

ebenolson, 2019-02-08:
Very similar problem affected me when I based my CM3 board on the CMIO schematic. 

Agree that this is not confusing but simply wrong - the schematics do not reflect the reality of the design, and a board manufactured according to them will not function.

aallan, 2021-06-08:
Schematics and other non-web content will be moved out of the repo ahead of #1911. Closing.

closed 2021-06-08 by aallan