raspberrypi/documentation #857: Headless wireless config explanation missing

z8i, 2018-03-18:
This blogpost https://www.raspberrypi.org/blog/another-update-raspbian/
explains how to set up wifi while creating a raspbian-sd-card.
I would have been happy, if i had found it somewhere here:
https://www.raspberrypi.org/documentation/configuration/wireless/README.md

So i like to propose to add this method to that wireless configuration methods.
"If a wpa_supplicant.conf file is placed into the /boot/ directory, this will be moved to the /etc/wpa_supplicant/ directory the next time the system is booted, overwriting the network settings; this allows a Wifi configuration to be preloaded onto a card from a Windows or other machine that can only see the boot partition."

JamesH65, 2018-03-19:
Proposed changes on PR #860

JamesH65, 2018-09-12:
Closed as docs updated.

closed 2018-09-12 by JamesH65