raspberrypi/documentation #866: B3+ Schematic - POE circuit

tonysmith55, 2018-03-23:
The B3+ schematic has been updated to show the POE circuits but unfortunately it shows the centre taps of the 4 Ethernet transformers connected together inside the Ethernet socket (J10).  As a result it implies all 4 pins of the new header (J14) are also joined. If you ignore the circuit diagram inside the Ethernet socket (J10), there is enough information from the terminal names to determine which pin of  J14 connect to each of the 4 Ethernet circuits. BTW, this was not obvious to start with, only realised I could figure it out some time later.   Also, no information either on Schematic or Mechanical drawing to determine the orientation of the connectors.  Tends to be obvious with other long rectangular connectors but not with the square connector J14.    https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi-3bplus_reduced_schematic.pdf

JamesH65, 2018-07-02:
Link has changed.  This is the new document. https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_3bplus_1p0_reduced.pdf

Roger-Thornton, 2018-07-05:
> @Roger-Thornton <https://github.com/Roger-Thornton> This is out of my
> area! Any comments?

The Schematics label each of the Taps on the header, TR0_TAP relates to
TR0n/p, the board marks the pin 1 with a square footprint. Pinout attached
for reference.

tonysmith55, 2018-07-05:
Hi Roger,  Not sure if you appreciate my comment does not relate to the schematic itself but inside the library function which is drawing the Ethernet connector and magnetics.  If you look to the right of the pins 11, 12, 13 & 14 there is only one vertical wire and this connects to all 4 transformer centre taps.  It looks like pins, 11, 12, 13 & 14 are connected to all 4 transformer centre taps.

I’m sorted, have commissioned a fully compliant POE power supply PCB but thinking of novices who could be confused by the diagram of the wiring inside the ethernet connector. 

Regards
Tony Smith

From: Roger Thornton
Sent: Thursday, 5 July 2018 6:45 PM
To: raspberrypi/documentation
Cc: tonysmith55; Author
Subject: Re: [raspberrypi/documentation] B3+ Schematic - POE circuit (#866)

The Schematics label each of the Taps on the header, TR0_TAP relates to
TR0n/p, the board marks the pin 1 with a square footprint. Pinout attached
for reference.

JamesH65, 2018-09-12:
@Roger-Thornton Do you think we need to make any alterations here?

JamesH65, 2018-10-01:
@Roger-Thornton Taken a look, the diagram is as described above, but, as a non-expert, I don't see an issue with it as it stands. I pressume this is a standard library function so assume it's the correct way of describing the device.  Using the external/interface labelling is the correct approach rather than looking at the device internals, and that is presumably correct. Inclined to close.

gpertea, 2018-10-17:
Confused novice here: we are actually trying to use an Adafruit PoE module and their instructions specify connecting the module to the **J14** pins while referring to those pins as _"Ethernet wires 4 and 5"_ and _"Ethernet wires 7 and 8"_. To our shame we have a very hard time, looking at the very schematic being discussed here, figuring out how those "Ethernet wires" actually connect to the **TRx_TAP** labeled pins of J14 in the schematic/diagram. So we'd be very grateful for any clarifying annotations or improvements of the current PoE documentation/schematic, as suggested in this topic.

JamesH65, 2018-10-18:
@Roger-Thornton Looking at the PCB there is no indication which pin is which on the silk screen, and since its square its quite awkward.

JamesH65, 2019-01-18:
Ping @Roger-Thornton

JamesH65, 2019-01-21:
OK, so pin one on the PoE header is indicated by a square pad on the PCB, so that easy enough to find (except I didnt first time round, but I'm not a HW engineer).  As for the comment from @gpertea, it seems fairly obvious to me that ethernet lines 4/5/ and 7/8 refer to the TR2 and TR3 taps.

The schematic representation of the jack is a little bit unclear, as it uses a single line to represent all 4 separate tap lines.

Cannot comment on the different uses of 0/1 vs 2/3 taps.

aallan, 2021-06-08:
We're moving schematics and other such things out of the repo as part of #1911. So closing.

closed 2021-06-08 by aallan