raspberrypi/documentation #869: Add section on the files int he boot folder

JamesH65, 2018-03-26:
Nothing in the docs describes what any of the files in the boot folder actually do.

Probably best to link a new page from https://www.raspberrypi.org/documentation/configuration/

lurch, 2018-03-27:
If it helps, there's a tiny bit of documentation I wrote about this quite a while ago at https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained

JamesH65, 2018-03-27:
That would be an excellent start if you don't mind me nicking some of it!

lurch, 2018-03-27:
Of course not, go for it :)

JamesH65, 2018-03-27:
First pass at this https://github.com/raspberrypi/documentation/blob/JamesH65-patch-boot-folder/configuration/boot_folder.md

lurch, 2018-03-28:
"In a basic Raspbian install the boot files are stored on the first partition of the SD card, which is VFAT formated."

Isn't the same thing true for **any** Raspberry Pi OS, not just Raspbian? ;-)

Probably wouldn't hurt to add a line about `kernel.img` (for Pi1) and `kernel7.img` (for Pi2 & Pi3) too?

JamesH65, 2018-03-28:
I was trying to avoid mentioning Noobs as I have no idea how that is all formatted. but presumably its the same, so will do as suggested.

Good idea on the kernel stuff too. Will try and update it today sometime.

lurch, 2018-03-28:
> I was trying to avoid mentioning Noobs...

That's not quite what I was thinking (I was thinking more along the lines of other RPi Linux distros, not just Raspbian), but yeah I guess it makes sense to try and be definitive and cover NOOBS too. The differences are explained here: https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained#noobs-bootup-low-level  (it's the same files, just renamed to different filenames)

XECDesign, 2018-03-28:
Other distros may throw u-boot and grub/uefi into the mix, which would complicate the description of all the files a fair bit. We can really only speak for Raspbian, IMHO.

lurch, 2018-03-28:
> Other distros may throw u-boot and grub/uefi into the mix

Good point, I hadn't considered that! :-/

lurch, 2018-05-31:
I guess this is closed by #883 ? :wink:

lurch, 2018-07-02:
ping @JamesH65 This can be closed too :smiley:

JamesH65, 2018-07-02:
Yup!

closed 2018-07-02 by JamesH65