raspberrypi/documentation #872: Module install cmd line has issues.

Dutchdoc, 2018-04-02:
Nowadays sudo sets a secure path (specified in the sudoers file), which means that the module install cmd line as specified in the instructions won't find the compilers and thus outputs some error messages.

Instead of 

`sudo make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=mnt/ext4 modules_install
`

use

`sudo env PATH=$PATH make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=mnt/ext4 modules_install
`

(Or, less safe, remove the "Defaults   secure_parh=" .. etc from the sudoers file).

lurch, 2018-04-03:
I strongly suspect that the instructions originally worked when written, so perhaps this is one of those awkward situations where some (newer) Linux distros work slightly differently to other (older) Linux distros? In which case, would making the changes you suggest stop the instructions working on distros where they _used_ to work? Or are the changes you're suggesting fully backwards-compatible? (if you see what I mean)

Dutchdoc, 2018-04-03:
I share your suspicion that this started to happen when the latest distros started to add the line "Defaults  secure_path=' <path>'" to their sudoers file.
I also think that the errors now showing up can be ignored: looks like, for whatever reason, only want to get the version of the compiler, nothing is really being built during 'install modules'.
But still .. there shouldn't be any errors.

If my suggestion is backward compatible, I don't know (I do assume so, though), and is it worth to find out? If we have a cmd line that ONLY works on  older systems, and one that ONLY works on newer systems, which one should we go with? I personally prefer using the latest tools and distros.

Fortunately, this is purely a documentation issue: meaning we can actually explain, that if the first cmd line yields errors, then use the 2nd one.

Or .. tell them to check their sudoers file for that 'Defaults  secure_path=""' line, and base their choice on that,
Or .. tell them to run this test: sudo arm-linux-gnueabihf-gcc --version, and if the cmd can't be found, use the 2nd form of the cmd line given.

All that needs to be done, is explain the two choices in the instructions, that's all ;-)

JamesH65, 2018-04-10:
If someone who knows would like to produce a PR that would be splendid.

lurch, 2019-09-07:
Fixed by #1261

closed 2019-09-07 by lurch