raspberrypi/documentation #873: demo docs duplication?

lurch, 2018-04-03:
I noticed that the recent blogpost https://www.raspberrypi.org/blog/raspberry-pi-easter-eggs/ links to https://projects.raspberrypi.org/en/projects/demo-programs which looks an _awful_ lot like https://github.com/raspberrypi/documentation/blob/master/usage/demos/README.md

IMHO duplication is bad because it means that fixes or improvements made in one location (usually) don't get made to the other location.
Should one-or-the-other be deleted and replaced with "Content moved to ..." redirect links?

bennuttall, 2018-04-03:
Actually that's the origin of the content, not the docs.

I copied the contents of Dave's demo programs resource (along with a few others) into the docs when creating the usage guides. I later replaced some of these with links to their resource URLs to avoid duplication, but I guess this one got left behind. I'll update when I get chance.

lurch, 2018-04-03:
Oh, my mistake then ;-) 
Any idea why e.g. https://projects.raspberrypi.org/en/projects/demo-programs/5 is missing the `cd` and `ls` commands? (which makes the "You’ll notice the `.bin` file is shown in green." text a bit confusing)

Also, it probably doesn't make sense for https://projects.raspberrypi.org/en/projects/demo-programs to be tagged with `Python` ?

bennuttall, 2018-04-03:
I'll pass that on

martinohanlon, 2018-04-04:
I have updated the demo-programs project.

JamesH65, 2019-01-18:
@bennuttall @martinohanlon Can this be closed?

martinohanlon, 2019-01-20:
I resolved the issue. I say yes.

lurch, 2019-01-20:
I think the original point still stands? There's both https://projects.raspberrypi.org/en/projects/demo-programs and https://www.raspberrypi.org/documentation/usage/demos/README.md which duplicate a lot of each other's content, e.g. https://projects.raspberrypi.org/en/projects/demo-programs/8 and https://www.raspberrypi.org/documentation/usage/demos/hello-fractal.md

@bennuttall 's previous content seems to suggest that the "projects" version is the original, so perhaps the versions in the "documentation" repo should be updated to point to the "projects" URLs?

martinohanlon, 2019-01-21:
Ok, I see...

I dont know the history behind the demo-programs either on the projects site or the in the documentation, but here are some comments.

https://projects.raspberrypi.org/en/projects/demo-programs isnt listed on the projects site, this just means it isn't "browsable" from projects.raspberrypi.org, so it will only be getting visits where it is directly linked too.

Its ok to have "content" duplicated on the projects site and the documentation - this is by design as they appeal to different audiences and often have very different structures (albeit for demo-programs its very similar). 

I would be tempted to leave as is, the entry points into this content is very different and is there to serve different audience, a context switch (i.e. leaving the documentation site to go to projects site) may cause confusion. 

@JamesH65 if you feel different and want to link out to the projects site, no prob, just let me know and I will make a note.

lurch, 2019-01-21:
> Its ok to have "content" duplicated on the projects site and the documentation - this is by design as they appeal to different audiences

> I would be tempted to leave as is

Fair enough - I suspected that might be the ultimate conclusion of this ticket... :wink: but doesn't hurt to check.

lurch, 2019-09-19:
Fixed by #1139

closed 2019-09-19 by lurch