raspberrypi/documentation #874: PI AP is not viewable on any device

melzoghbi, 2018-04-04:
Hi,

I am working to setup RPI as AP as specific in this [article](https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md). When i followed all instructions, and rebooted the RPI AP is not accessible or viewable on my iPhone or PC so i can connect through the wifi. 

I am implementing this on a PI Zero W loaded with latest raspbian image.

Let me know what i am missing so i can see the AP wifi network i set on the RPI device.

Thanks.

bennuttall, 2018-04-04:
Sorry, this is the wrong place to be reporting user issues.

This is for reporting issues with the documentation contents, not troubleshooting issues. Please post your question on the forums, and if you find that some key information is missing, or something in the docs is missing, please open an issue here then.

Thanks

closed 2018-04-04 by bennuttall

melzoghbi, 2018-04-05:
Hi @bennuttall,

The documentation is not accurate since it doesn't do what is suppose to do for Raspbian Stretch OS (march version). I tried the same instructions on RPI3 Model B and RPI Zero W and i was not be able to configure an AP for my PI.

bennuttall, 2018-04-05:
@JamesH65 @XECDesign any ideas?