raspberrypi/documentation #877: Electrical Characteristics missing details for VDD_IO=3.3V

6by9, 2018-04-09:
As noted in forum thread https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=138215, the DC electrical characteristics quoted in the Compute Module datasheet are for VDD_IO at 1.8V and 2.7V. The Pi is running VDD_IO at 3.3V, so the details are missing.
Can we add those details?

And are there any details about the level of hysteresis (it's always enabled by the firmware with no supported mechanism to disable it, so it is the appropriate mode of operation to be quoting values against).

Ping @Roger-Thornton.

Roger-Thornton, 2018-04-10:
@mike-stimson will be characterising this at some point.

JamesH65, 2018-07-02:
Fixed. Now on the GPIO page https://www.raspberrypi.org/documentation/hardware/raspberrypi/gpio/README.md

lurch, 2018-07-02:
The `Output high current` section lists `VDD IO = 2.7V` twice - presumably the last one was supposed to be `VDD IO = 3.3V` ? :mag:

JamesH65, 2018-07-03:
I thought I had fixed that.....maybe that was in the PDF, anyway, good spot. I'll fix it today.

JamesH65, 2018-07-03:
Ah, already done. Thanks.

closed 2018-07-03 by JamesH65

seamusdemora, 2022-02-17:
Perhaps I have missed this, but I cannot find a definition of `VDD IO`. I have searched for it in the [online documentation](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#voltage-specifications), the [Compute Module datasheet](https://www.raspberrypi.com/documentation/computers/compute-module.html#datasheets-and-schematics), and the PDF [Compute Module 4 Datasheet](https://datasheets.raspberrypi.com/cm4/cm4-datasheet.pdf). 

If I've missed it, please provide a link; if not, could someone add a brief note that defines `VDD IO`? 

Thank you

seamusdemora, 2022-02-17:
FWIW: I also followed the resolution of #1463 & #2081 to [these changes](https://github.com/raspberrypi/documentation/pull/2081/files). There is a change: 

>NOTE: On the Compute Module devices, it is possible to change the VDD IO from the standard 3V3. In this case, V~OL~ and V~OH~ will change according to the table on the linked page. 

That **helps**, but a similar NOTE is needed for VIL and VIH below the `Voltage Specifications` table - where `VDD IO` is repeatedly referenced.  In the PDF [Compute Module 4 Datasheet](https://datasheets.raspberrypi.com/cm4/cm4-datasheet.pdf) I **do see** several references to pins (e.g. `GPIO_VREF`) which _**apparently means**_ the CM4 can use different voltage levels for the GPIO, but again - no mention of `VDD IO`. The NOTE above is the only thing I found that ties `VDD IO` to the GPIO supply rail voltage. 

Also, if these details that apply only to CM4, wouldn't it be simpler to just state that explicitly in the online documentation & refer the CM4 parties to the PDF?? 

Finally: I'll be happy to submit a pull request for this **_IF_** someone will confirm that `VDD IO` is in reality the GPIO supply rail.

lurch, 2022-02-17:
ping @dp111

seamusdemora, 2022-02-24:
@dp111 , @lurch :
> ping @dp111

I'd like to try to get this resolved - mostly because I think it should be clarified, but selfishly so that I can remove it from my "To Do" list. Again, I'll be glad to take a stab at writing a pull request, if someone will confirm that  `VDD IO` is in reality the GPIO supply rail.

dp111, 2022-02-24:
For the compute modules VDDIO is VDD_GPIO[xx-yy]  . For CM4 there is no
VDDIO , but GPIO_VREF is effectively the same thing., but as CM4 and CM3
are different products you shouldn't interchange specifications.

seamusdemora, 2022-02-25:
@dp111 : 

I'll try to parse that shortly, but in the meantime: what about some clarification in the documentation?

seamusdemora, 2022-02-25:
@dp111 : 

Just so we are clear: the [online documentation](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#voltage-specifications) repeatedly refers to a `VDD IO` parameter in the table. The table is not specific to any CM AFAIK, but is provided as documentation for all RPi in general. The fact that the table was ***extracted*** from a CM document is all well & good, but the term `VDD IO` does need some clarification in the documentation. Don't you agree?

lurch, 2022-02-25:
@seamusdemora I'm still not sure what it is you're asking. Do the [schematics](https://www.raspberrypi.com/documentation/computers/compute-module.html#datasheets-and-schematics) for the CM3IO and CM4IO boards help?

seamusdemora, 2022-02-25:
@lurch : 

> @seamusdemora I'm still not sure what it is you're asking. Do the [schematics](https://www.raspberrypi.com/documentation/computers/compute-module.html#datasheets-and-schematics) for the CM3IO and CM4IO boards help?

OK - let me try again: 

1. The "Voltage Specifications" for all RPi GPIO are located at this URL:  https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#voltage-specifications 

2. If you review the data in the column labeled as `Conditions` in this table, you will see repeated references to a voltage called `VDD IO`. It takes on one of 3 values: 1.8V, 2.7V or 3.3V. 

3. Now, let's assume someone (like me :) has entered this table in an effort to learn what voltage limits constitute a logic "HI/1", and what voltage limits constitute a logic "LO/0" - i.e. what are `VIH` and `VIL`? In particular, I want to know what level of input voltage at a GPIO pin is required to ensure that it will result in a binary `0`. 

4.  To get this information from the "Voltage Specifications" table, I go to the row labeled `VIL`. But instead of a single value, there are three rows with three different values. I see that the information I need is ***dependent*** on the value of `VDD IO`. For `VDD IO = 1.8V`, VIL = 0.6V; for `VDD IO = 2.7V`, VIL = 0.8V; and for `VDD IO = 3.3V`, VIL = 0.9V. 

5.  Therefore, the answer to my question requires me to know what is `VDD IO`??? 

Again, I have searched the documents cited previously, but found nothing. I ***think*** that for all RPi `VDD IO` is the `3v3` voltage rail used for all GPIO, but I do not ***know*** this if it is not documented. All I am trying to get is an *acknowledgement* from someone who knows, that `VDD IO` is 3.3 Volts on the RPi - or if not, then what is it?  

A NOTE at the bottom of the "Voltage Specifications" table is needed to make this clear. I cannot resolve my question definitively if I do not know what `VDD IO` means for my Rapberry Pi 4B.  I realize that the CM hardware is different than the RPi hardware, and that it may be possible to set the GPIO supply rail to voltages other than 3.3V, but I don't use CM hardware. 

To make this clear, and **assuming** that `VDD IO` is in fact a constant 3.3V on all RPi except the CM models, I would propose adding NOTE **`d`** to read as follows: 

>d `VDD IO` is 3.3V for all Raspberry Pi models except for the Compute Modules, which allow the user to set `VDD IO` to 1.8V, 2.7V or 3.3V.

dp111, 2022-02-26:
For pi 1 2 3, VDDIO is 3.3v and you can use the table. For pi4B you should use the CM4 datasheet with GPIO_VREF set to 3.3v . This section of the page was created before the Pi4 was created . So does need some changes .

aallan, 2022-03-03:
See #2461 for a PR to resolve this issue.

seamusdemora, 2024-01-19:
Thank you @aallan !

(catching up on saved notifications...)