raspberrypi/documentation #881: Move FAQ in to the docs github repo.

JamesH65, 2018-04-11:
Currently the FAQ is in Wordpress, will be much easier to deal with if moved in to the docs github and auto updated to the website from there.

JamesH65, 2018-04-11:
There is a branch here that has started this...

https://github.com/raspberrypi/documentation/tree/faqs

JamesH65, 2018-07-02:
@bennuttall I think this can now go live. Not had any negative response to the move. @LizUpton Any thoughts?

JamesH65, 2018-08-08:
Update. Current FAQ has been moved to github, and joined by a technical FAQ. These are currently being merged, andthe original FAQ is being trimmed to remove out of date entries.

JaninaPi, 2018-09-05:
I have done the merging and trimming and @LizUpton has signed the changes off: https://github.com/raspberrypi/documentation/blob/master/faqs/README.md
However, in the process I've messed up some of the internal links: the ToC headings from 'Performance' through to 'SD cards', as well as 'Troubleshooting', don't go to the main text, and I can't figure out why. It must be something really easy to fix, but I can't spot the problem.

bennuttall, 2018-09-05:
Oh that's because GitHub makes its own anchor links based on header text. The version on our site doesn't do that. But to work on both versions, make sure links are unique, i.e. not the exact same as the header.

JamesH65, 2018-09-05:
IIRC, you need to set the link text to exactly match the header text, so if you changed the header you need to change the link to match.

JamesH65, 2018-09-14:
Where are we with this? Would be good to get it live.

JaninaPi, 2018-09-14:
I've changed the header links now so that they do not match the heading text exactly, but do match the anchors, so at least on GH, all of them do send you to the right section now. So this should be all good to go.

bennuttall, 2018-09-17:
Ping @dtt101

dtt101, 2018-09-17:
Just let me know what needs to be updated in the footer. Can we just update the copy on the old site with a message?

JamesH65, 2018-09-17:
This sort of message "This has moved to here"?

The link to change is in the footer, FAQ, which will need to now link to this new document at "documentation/faqs/README.md"

bennuttall, 2018-09-21:
Actually we don't need the footer changing, just a redirect adding:

https://www.raspberrypi.org/help/faqs/ -> https://www.raspberrypi.org/documentation/faqs/

bennuttall, 2018-09-21:
I mean the link in the footer can change to be the canonical one, but that's not vital once the redirect's in place.

damienh, 2018-09-21:
@bennuttall @JamesH65 I have done the redirection via the wordpress plugin. I will leave this issue open so that it can eventually be changed in the footer itself, but for now its ok.

JamesH65, 2018-09-24:
Now live, and the redirection means the footer goes straight to the new page, so closing since the footer change is actually not a docs issue.

closed 2018-09-24 by JamesH65