raspberrypi/documentation #886: Add a section on the OTP bit values

JamesH65, 2018-04-16:
See https://github.com/raspberrypi/firmware/issues/974 for details.

JamesH65, 2018-07-19:
See https://github.com/raspberrypi/documentation/blob/JamesH65-otpbits/hardware/raspberrypi/otpbits.md for a WIP

lurch, 2018-07-24:
I wonder if "All member of the Raspberry Pi range have a onboard One Time Programmable (OTP) memory block." would be better phrased something like "All SoCs used by the Raspberry Pi range have an inbuilt One Time Programmable (OTP) memory block." ?

To me, "onboard" makes it sound like it's just another chip on the PCB, rather than something that's part of the SoC itself, IYSWIM.

JamesH65, 2018-07-24:
Makes sense, I'll make some changes.

JamesH65, 2018-08-08:
This is now merged

https://www.raspberrypi.org/documentation/hardware/raspberrypi/otpbits.md

closed 2018-08-08 by JamesH65