raspberrypi/documentation #902: RPi Zero W OTG Schematic

jebentancour, 2018-05-09:
I can´t find the schematics of this part. I want to know if it its possible to use USB OTG and power the Raspberry Pi from the GPIO header.

lurch, 2018-05-15:
AFAIK all the 5V rails on the Pi Zero W are simply connected together; which means if you want to use it in USB OTG mode (i.e. having the PiZero appearing as a USB device instead of a USB host) then you either need to prevent the Pi also being powered via the GPIO header _or_ you need to snip the 5V wire in the USB OTG cable.

JamesH65, 2018-05-21:
The schematics are not released, so probably best to ask this sort of question on the forum, although @lurch has probably [provided the information needed.

closed 2018-05-21 by JamesH65

jebentancour, 2018-05-22:
Thanks @lurch !