raspberrypi/documentation #903: recommended etcher tools fails - please do not recommend

COLABORATI, 2018-05-12:
The recommended "etcher" tool does not work on Ubuntu 18.04 with XFCE. It would be great to remove that recommendation as the team does not seem to test the software before releasing and recommend something that works. In the first place it is better to teach users what really happens behind the scenes, what you are doing great with the detailed explanation docs!

JamesH65, 2018-05-12:
The docs already include a command line option for imaging SD cards, so there is already a workaround if Etcher fails to work for some reason. However, since Etcher is multi platform and works on most systems (and presumably will be fixed to work on failing system fairly quickly), its still the best options for the majority of users. I, for example, have never had any problems with it.

lurch, 2018-05-14:
ping @jhermsmeier

jhermsmeier, 2018-05-14:
Thanks for the ping @lurch. @COLABORATI while we do not test on every OS / distro released under the sun, we've been aware of the particular issue you're talking about, and will fix it.

JamesH65, 2018-05-14:
Thanks all, I'm happy with our documentation at this point, will close this issue.

closed 2018-05-14 by JamesH65