raspberrypi/documentation #907: Change password when enabling remote access

aldreth, 2018-05-18:
https://github.com/raspberrypi/documentation/blob/master/remote-access/ssh/README.md & https://github.com/raspberrypi/documentation/tree/master/remote-access/vnc both note that:

> As of the November 2016 release, Raspbian has the SSH server disabled by default.
(or similar)

There are instructions on how to enable these interfaces, but no note that it might be a good idea to change the password for `pi` or how to do that.

lurch, 2018-05-20:
Good idea, I guess it just needs a link to https://github.com/raspberrypi/documentation/blob/master/configuration/security.md

JamesH65, 2018-05-21:
Good idea, I'll try and implement it today.

JamesH65, 2018-05-21:
See #909.

Didn't see anywhere that needed updating on the VNC page though? Is there something in particular that need changing on that one?

aldreth, 2018-05-22:
Doesn't VNC use the same password as ssh? It certainly seemed to when I was setting it up?

So if you set up VNC but not SSH, you should still change the password I assume.

JamesH65, 2018-05-22:
No expert on VNC, but from reading the docs I think it uses a different password. There certainly no relevant references to SSH on the VNC page.

aldreth, 2018-05-22:
Well I didn't even read the docs @JamesH65, so I imagine you're right!

JamesH65, 2018-07-03:
Merged

closed 2018-07-03 by JamesH65