raspberrypi/documentation #911: reboot needed before masquerade on pi zero w March 2018 version

midol, 2018-05-26:
right after:

Add routing and masquerade

Edit /etc/sysctl.conf and uncomment this line:

net.ipv4.ip_forward=1

and before the masquerade code is added a reboot is needed so that the /lib/modules will line up with the modified kernel version. I spent a lot of time on this.
email me - geek@uniserve.om

JamesH65, 2018-05-27:
Can you provide a link to the page where this needs to be updated?

lurch, 2018-05-31:
@JamesH65 this seems to be referring to https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md#add-routing-and-masquerade ?

JamesH65, 2018-07-03:
@midol I'm a bit confused as to why you need to realign the modules with the updated kernel - what updated kernel?

JamesH65, 2019-01-18:
@midol Any comment on my previous question?

closed 2020-11-19 by JamesH65