raspberrypi/documentation #912: Missed eject the card step for Mac OS / Command line install instruction

aleksejleonov, 2018-05-30:
https://github.com/raspberrypi/documentation/blob/master/installation/installing-images/mac.md

**Alternative method** section have *eject the card* step in instruction:

```Markdown
- After the `dd` command finishes, eject the card:

    ```
    sudo diskutil eject /dev/rdisk3
    ```

    Alternatively, open Disk Utility and use this to eject the SD card.
```

This step missed in **Command line** section:

```Markdown    
    - If this command still fails, try using `disk` instead of `rdisk`, for example:
    
       ```
       sudo dd bs=1m if=2018-04-18-raspbian-stretch.img of=/dev/disk4 conv=sync
       ```
       or
       ```
       sudo dd bs=1M if=2018-04-18-raspbian-stretch.img of=/dev/disk4 conv=sync
       ```

## Alternative method
```

I believe, **Command line** section ending should be:
```Markdown
- After the `dd` command finishes, eject the card:

    ```
    sudo diskutil eject /dev/rdisk<disk# from diskutil>
    ```

## Alternative method
```

lurch, 2018-05-31:
From what I can tell (I don't have a Mac myself), I think you only need to "software eject" a card (i.e. ensure the buffers are fully written to disk) if using e.g. `/dev/disk4`, and not if using `/dev/rdisk4` ?

https://github.com/raspberrypi/documentation/issues/693#issuecomment-315331670

aleksejleonov, 2018-05-31:
@lurch, you are right, looks like eject not required when using `/dev/rdiskN`.

But **Command line** section contains both ways, with `/dev/diskN` and `/dev/rdiskN`.
> If this command still fails, try using disk instead of rdisk, for example

So, adding *eject the card* step could help someone using `/dev/diskN` instead `/dev/rdiskN`.

JamesH65, 2018-07-03:
Proposed fix in #934