raspberrypi/documentation #913: AIY-Projects - Vision Kit 1.1 - MP4box - sudo apt-get install -y gpac not working

davesdere, 2018-05-31:
I used it few months ago but now on a fresh install. It doesn't work anymore.
It returns
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package gpac

davesdere, 2018-05-31:
It works now.

closed 2018-05-31 by davesdere

lurch, 2018-05-31:
You probably forgot to run `sudo apt-get update` first?

davesdere, 2018-05-31:
I actually did
`sudo apt-get update`
`sudo apt-get upgrade`  then something broke.
I did `sudo apt-get update` again and something was still broken.
So I did `sudo apt --fix-broken install` and `sudo apt-get upgrade`.
Then `sudo rpi-update`. I also installed bunch of libraries while looking for a fix.
Reboot. Then worked on something else. Then it was really bothering me so I tried again and it worked.
Could it be possible that it was only due to a timeout request on a unresponsive server while looking for the package?

lurch, 2018-05-31:
Who knows - the internals of apt tends to be a bit of a black box ;-)