raspberrypi/documentation #920: Compute Module Datasheet Type-O?

gudenau, 2018-06-10:
On page 11 of the Compute Module datasheet there appears to be a type-o. It says `USB_DM` where it appears as though it should be `USB_DN` for USB Data Negative.

[Link to the document.](https://www.raspberrypi.org/documentation/hardware/computemodule/datasheets/rpi_DATA_CM_1p0.pdf)

[image]

JamesH65, 2018-06-11:
@jimbojr @Roger-Thornton Hi guys, any thoughts on this?

Roger-Thornton, 2018-06-11:
DM or DN are both used for negative part of a differential pair labelling,
both are fine. As we've used DN elsewhere we could change to be consistent
though.

lurch, 2018-06-11:
I'm not a hardware engineer myself, but I'd always assumed (guessed?) it was D+/D- or DataPositive/DataNegative or DataPlus/DataMinus as all being fairly interchangeable?

JamesH65, 2018-06-14:
Closing this issue as questions answered/issue resolved.

closed 2018-06-14 by JamesH65