raspberrypi/documentation #924: Using the Raspberry Pi as an access point to share an internet connection (bridge)

kelsieflynn, 2018-06-17:
Regarding the work about bridge mode configuration, from here:
https://github.com/raspberrypi/documentation/blob/master/configuration/wireless/access-point.md

I have noticed a problem that seems to be recurrent where the sysv daemon at /etc/init.d/hostapd
does not always start correctly when in bridge mode. Example, on the very first bridge insertion attempt and start, it starts but exits without forking the sub processes and then exists normally.
The result of which is the wlan0 or wlanX interface doesn't get added to a bridge and no AP.

I have found that repeated service restarts will sometimes resolve the first startup issue and then it works afterwards as designed.
However, a systemd service instead of the /etc/init.d/hostapd appears to work everytime as designed.

Please consider updating hostapd to include a hostapd.service in systemd. I have a preliminary
one I use that can be found in my fork.

https://github.com/kelsieflynn/raspberrypi-documentation/blob/master/configuration/wireless/access-point.md

If you are not ready for a hostapd.service with systemd, please have a engineer consider what I have 
tried to explain.

Finally, I have no problem 'per say' and just wish to share the information I have. You may use this or close and ignore this issue, if you disagree in anyway.

lurch, 2018-06-18:
ping @XECDesign 
Is this an issue for https://github.com/RPi-Distro/pi-gen ? Or is it more of an upstream Debian thing?

XECDesign, 2018-06-18:
Not a pi-gen issue, but somebody would need to set aside some time to look into it to figure out what it is exactly. I'm not putting my hand up just yet, but i'll check if there's anything obvious next time I'm working with hostapd.

kelsieflynn, 2018-06-18:
Thanks for considering looking into this when you get the time.  I use hostapd with pi's extensively. They work well for me as I only need low power/short distances. I rarely use a standalone wireless router anymore.

More about the sysv hostapd "problem"

I have a tinkerboard(another arm7hf) that did the same thing until after the first start. Then it works okay. 
I also have a galliumos 2.1 box (x86_64, ubuntu xenial 16.04.X)   that I tried bridging on using the same method. It uses a different build as you may now than the deb version on rpi/tinker, but same version.

The differences I see is that ubuntu doesnt use dhcpcd, as it uses NetworkManager and different startup services. The ubuntu xenial works the same with a sysv init script controlling it. Same problems occur with the sysv init, in my experiences.

From what I could tell, even on the version with ubuntu xenial, when used in bridging mode, the sysv method is flacky. Out of 10 consecutive restarts,  at times it just fails silently(as in exited 0). Once I got a normal error:
"systemd: hostapd.service : Start request repeated too quickly"

After this I tried a new angle and discovered something that should give some clues.
On restarts, the sysv init version acts like a toggle.
Example: 
It would alternate between actually restarting and disabling on each consecutive run: 
````
systemctl restart hostapd

````
Some may say this is by design with the sysv hostapd init, but I would disagree, and that it shouldnt toggle regardless of the last state previously. Eg. systemd
restarts it regardless of the last state, with no toggling. 
* Also note: I modified the sysv stanza for restart @ sleep 'N' between stop/start, but this had little effect. It still toggled between 0/1 on each pass or gave a error that I was restarting too fast.
	

	
In comparison to all(rpi/tinker and galliumos(x86_64)) , they all worked properly(bridging) when I manually started hostapd with
````
hostapd /etc/hostapd/hostapd.conf -B -P /run/hostapd.pid
````

I tried a systemd hostapd.service and the flacky mis-starts disappeared. 
For a reproducible scenario: 
one could run:

````
watch -n5 systemctl restart hostapd

````

You should see the toggles.

and with the systemd version, not one single error. Restarts as intended with no problems shown.

I dont claim to know the underlying problem, just presenting the high spots. I generally would speculate further on why, but in this case, I dont know all the development changes between hostapd/systemd to do it properly, so I'll leave the advanced speculation for someone else to do.

JamesH65, 2019-01-18:
Ping @XECDesign

XECDesign, 2019-01-22:
I think the simplest thing to do would be to steal hostapd.service from the buster package, if @kelsieflynn is still around to test that it resolves the issue. https://salsa.debian.org/debian/wpa/raw/debian/2%252.6-20/debian/hostapd.service

Then the issue will resolve itself in Buster.

kelsieflynn, 2019-01-23:
Seems to be working with that service file just fine.

https://salsa.debian.org/debian/wpa/raw/debian/2%252.6-20/debian/hostapd.service

I had muddled my setup quite a bit so to get a clean start :

I backed up my hostapd.conf file
	systemctl stop hostapd
	Removed my old testing hostapd.service from /etc/systemd/system
	Upgraded to lastest pkgs with apt-get
	Removed/purged old hostapd
	Reinstalled hostapd/ added back my custom hostapd.conf
	dropped the buster hostapd.service from above URL into  /etc/systemd/system
	systemctl daemon-reload
	systemctl enable hostapd
	systemctl start hostapd

systemctl -l status hostapd
Output:

root@raspberrypi:/etc/systemd/system# systemctl -l status hostapd
● hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
   Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-01-23 06:00:20 EST; 24s ago
  Process: 25025 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=0/SUCCESS)
 Main PID: 25026 (hostapd)
   CGroup: /system.slice/hostapd.service
           └─25026 /usr/sbin/hostapd -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf

Jan 23 06:00:20 raspberrypi hostapd[25025]: wlan0: AP-ENABLED
Jan 23 06:00:20 raspberrypi systemd[1]: Started Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
Jan 23 06:00:21 raspberrypi hostapd[25026]: wlan0: STA b8:XXXXXXXX IEEE 802.11: authenticated

*Not shown in output, but i also tested the DAEMON_OPTS by adding "-ddtK" to /etc/default/hostapd  and works fine too.

	
Rebooted,
restarted the service a few times, works.  Connected clients, checked with hostapd_cli, no problems.
 This buster one is very similar in content to the temporarily one i was using except i had manual pathing, So I'm  fairly confident its a stable workaround.

  
Thank you.

XECDesign, 2019-01-23:
Thanks, and sorry it took this long. I'll update the package some time this week.

kelsieflynn, 2019-01-23:
Np, great!

XECDesign, 2019-03-05:
The buster version of wpa/hostapd have been backported to stretch. I don't know if the documentation needs an update or if this is okay to close.

kelsieflynn, 2019-03-06:
Yes, close this.
Thank you.

closed 2019-03-06 by kelsieflynn