raspberrypi/documentation #929: SSH on Windows

LunarWatcher, 2018-06-30:
Specifically talking about [the Raspberry Pi docs][rpidoc] (emphasis mine):

> On Windows you will **need** to download an SSH client.

Windows 10 (at least, not tested on other versions) supports `ssh` like UNIX and Mac. See [this][1] and [this][powershell]

Also, [cygwin][cyg] can come with OpenSSH too. AFAIK, it works on Win7 and 8 as well. Which means, if OpenSSH is installed (which does come with Cygwin or since the fall creators update 2017), there's no need to install an SSH client. 

[1]: https://github.com/PowerShell/Win32-OpenSSH/wiki/Install-Win32-OpenSSH
[cyg]: https://superuser.com/a/1111661/623128
[powershell]: https://blogs.msdn.microsoft.com/powershell/2017/12/15/using-the-openssh-beta-in-windows-10-fall-creators-update-and-windows-server-1709/
[rpidoc]:  https://www.raspberrypi.org/documentation/remote-access/ssh/windows.md

JamesH65, 2018-07-02:
So there is a command line SSH install on my Win10 box, not sure whether its been there from the start or whether its been installed as a result of something else. I going to hazard a guess that if someone has installed OpenSSH then they are probably savvy enough to understand about SSH clients already, so only a slight change to the docs might be necessary i.e. "Depending on your version of windows, you may need to install a SSH client" or similar. We'll still recommend putty I think.

lurch, 2018-07-02:
Also worth noting that the `Install-Win32-OpenSSH` wiki link that @LunarWatcher provides above refers to installing the SSH *server*, which obviously isn't required if all you're doing is connecting to a Raspberry Pi!

JamesH65, 2018-07-03:
OK, will make some minor changes as I outlined above.

JamesH65, 2018-07-03:
Proposed change in #933

lurch, 2018-07-03:
@LunarWatcher #933 has been merged - are you happy for this to be closed?

bsimmo, 2018-07-12:
Just an fyi.
The April 2018 Update has the OpenSSH install by default.
Before that it was still in testing and had to be installed by yourself.
 
It should behave like on Linux from the commandline.  sftp and scp are there to.
So anyone on that version can just follow the standard versions.

Of course the Ubuntu Linux subsystem could also be used on Windows10 as another method

A universal-ish method would be to use Google Chrome and their Secure Shell Extension (as used/shown by them with the Google AIY products).

lurch, 2018-07-12:
> A universal-ish method would be to use Google Chrome and their Secure Shell Extension (as used/shown by them with the Google AIY products).

IMHO "download and run putty" is much easier than "download and install Google Chrome, and then install the Secure Shell Extension for Chrome" :wink:

JamesH65, 2018-07-12:
I've happy with what it currently says (#933), so going to close. Please comment or provide a PR if you feel there are still problems.

closed 2018-07-12 by JamesH65

bsimmo, 2018-07-12:
Putty is nice, but it needs elevated privileges to run it unless they unblock (right click - properties - at the bottom, click unblock)

It would be nice to tell the user how to use the built-in Windows10 ssh utility, not just dismiss it.
Right click on Start Menu, select Command Prompt*
Type 
ssh pi@raspberrypi
etc.

if it doesn't come up with 
usage: shh [... etc help info stuff]
then it is not installed
 
So then suggest using putty as an alternative (or update windows)

*among many of the methods to get teh command line

lurch, 2018-07-12:
@bsimmo I'm sure @JamesH65 would welcome feedback / suggestions on #933 ...

(although I'm not sure if there are enough Windows users specifically running the April 2018 Update of Windows 10 to suggest those as the default instructions? I suspect there's still lots of people on Windows 7 and Windows 8 ?)

JamesH65, 2018-07-12:
I'm not sure how much instruction we should be giving to using Windows specific packages, especially ones that change without us knowing. Putty hasn't really changed for about 20000 years. These packages shoud provide their own instructions.

bsimmo, 2018-07-12:
It's the built in package, over 50% of Win10 users were using it a month after it's release and it's been a few month since. So the majority (a big enough majority to make Brexit happen!)

As for Win7/8/8.1 users, that is trying supporting old setups, it would be like you supporting Wheezie and Jessie ;-).

But it is why I worded it like I did.
The first tests it is there (you then follow the same Linux instructions).
It's a simple test two mouse clicks and four keyboard strokes.

After that it's just follow the same putty instructions you already have.

I don't mind adding the instructions.

JamesH65, 2018-07-13:
I'm happy to check out any PR's - I just don't have time to do this myself